hawku/TabletDriver #432: Auto restart

xXgolderXx, 2019-06-21:
Driver doesn't auto restart if PC is powered on from sleep mode and the tablet area get's set to full area instead of the given configuration

## Hardware/software information
Windows Version: 10 64 bit
Tablet Driver version: 0.2.3
Tablet model: XP-Pen G640

LegitDiamond, 2019-06-21:
I have the same issue with the same tablet. As for the tablet becoming full area, I believe that is because the G640 has a feature in which it will run in full area mode if it doesn't detect any drivers on the computer. So, when the hawku driver disconnects, the tablet will just switch to this feature.

xXgolderXx, 2019-08-20:
I noticed that after some time running idle (using mouse), the same thing happens, the tablet area get's set to full area instead of the given configuration. Does the driver disconnects after some time?

InfinityGhost, 2019-08-20:
Try a fresh install altogether to see if this issue still occurs. Use the newest release, or one of the newest releases from [my fork](https://github.com/InfinityGhost/TabletDriver/releases).

xXgolderXx, 2019-08-21:
Ok so it seems that there is a [pull request](https://github.com/hawku/TabletDriver/pull/382) to fix this issue but for some reason it's still open and your fork also doesn't have the change

InfinityGhost, 2019-08-21:
Wow I don't know how I missed that, going to pull that into a quick hotfix lol