hawku/TabletDriver #440: Cursor slows down depending on what window the mouse is hovering over

KuroUsagi962, 2019-07-07:
I have encountered a very strange issue with these drivers and don't really know what's going on.

To give some context as to how this might of happened, I was planning on using my tablet on my Mac but i wanted to get the exact same tablet area to use with the Mac Wacom drivers, I don't know of a way to extract/convert TabletDriverGUI's tablet area back to Wacom, so I installed the Wacom drivers and started manually editing the Wacom .tabletprefs file and importing it into TabletDriverGUI until I got the right values, I then uninstalled Wacom drivers and double checked that TabletDriverGUI was still working fine and it seemed to be.

Now I normally use Devocubs fork of this driver as it has Antichatter on it but I noticed that the cursor slows down depending on what its currently over, for example hovering over the TabletDriverGUI window cause it to feel very slow and sticky, this is the same when hovering over explorer.exe as well as the task bar, but hovering over a different program like discord makes it function like normal, it even functions normally whilst in the game osu! (fullscreen)

My first thought was that perhaps devocubs driver had a bug relating to Anti-chatter so I grabbed the latest version of this TabletDriverGUI and started it but it had the same issue, after fiddling around I found that turning off the Smoothing Filter fixed it completely. Another thing I found is that the issue seems less noticeable the lower the "Rate" is on the smoothing filter but even on the lowest rate the mouse appears as if the frame rate has been halfed/choppy.

Please let me know if my explanation was bad, I can record a video demonstrating it if you need it.

My only guess is that perhaps its something to do with the fact that I installed and uninstalled Wacom drivers, however I have deleted any excess files related to Wacom drivers and have re-installed vmmulti/TabletDriverGUI multiple times.

I've attached any files I think might be helpful, let me know if theres anything else you might need, thanks.

[files.zip](https://github.com/hawku/TabletDriver/files/3365762/files.zip)

## Hardware/software information
Windows Version: Windows 10 1903 Build 18362.207

Tablet Driver version: 0.2.3 Latest

Tablet model: Wacom Bamboo CTH-470

InfinityGhost, 2019-07-07:
I don't necessarily think this is an issue relating to anything you've done, its more than likely something already existing in the smoothing filter routine. I'm going to do some debugging and see what I can get. A video reference would help me out a lot in terms of reproducing the issue

InfinityGhost, 2019-07-07:
This issue may have already been fixed in the current master branch, so I'll be looking there first.

KuroUsagi962, 2019-07-07:
I just went to turn on NVIDIA overlay to make a recording, and it was giving me an error and I couldn't boot GeForce Experience, had to re-install it and now I can't reproduce the issue, weird....

Sorry for the waste of time!