hawku/TabletDriver #443: Driver recognizes Huion h610 as Huion osu! tablet.

HeyItsKairu, 2019-07-09:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: Huion H610

#255 
Same scenario as this, but I am using a Huion H610 rather than a Huion K58.
I know my tablet is not supported, and if adding support is not something you are interested in I understand.

InfinityGhost, 2019-07-09:
Add `getstrings 1 255` to the custom commands so we can find some descripting information about the tablet.

HeyItsKairu, 2019-07-09:
I managed to fix my problem by manually changing the Huion osu! tablet in tablet.cfg to fit correctly with the H610. If you would still like me to troubleshoot let me know, otherwise I can close the issue here. If you would like me to troubleshoot I added the command, but I don't know what to do with it.

InfinityGhost, 2019-07-09:
Which settings did you use for the H610? This will help add this tablet to this repo (eventually) and my fork

HeyItsKairu, 2019-07-09:
#
# Huion osu!tablet
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
CheckString 200 "HVAN"
Name "Huion osu!tablet (check the GitHub issue #99)"
ReportId 0x07
ReportLength 8
DetectMask 0x80
MaxX 20850
MaxY 12510
MaxPressure 2047
Width 254
Height 152.4
InitStrings 100 200

I won't claim to have done it perfectly but it appears to be working flawlessly for me.
(I'm guessing this is what you were asking for)

closed 2019-07-09 by HeyItsKairu