hawku/TabletDriver #448: Driver tells me to kill XP pen drivers even though they are already uninstalled?

Kiram123, 2019-07-19:
So I just got the Gaomon s620 (I already downloaded the new version of the drivers that support the tablet) and completely removed the drivers of the XP pen tablet I had before and removed Vmulti Drivers. I restarted my computer, and installed the Huion.bat file and started the driver. But it keeps repeating this in the console: 
1- Install Vmulti Driver
2- Kill Pentabletservice.exe(XP Pen driver)
3- Uninstall other drivers and reinstall Vmulti drive. 

## Hardware/software information
Windows Version: 

Tablet Driver version: 0.2.3

Tablet model: Gaomon s620

[image]

xXgolderXx, 2019-07-29:
I think you need to have vmulti driver installed no mater what tablet you're using and if you're using a Huion or Gaomon tablet you also need to have the huion drivers installed

RevENGE-code, 2019-09-02:
heres the solution https://www.reddit.com/r/osugame/comments/cywlbt/how_to_install_hawku_driver_for_the_gaomon_s620/