hawku/TabletDriver #454: [Not Important] Disable auto restart

HilmTien, 2019-08-06:
Ok, so this isnt very important but every time my tablet is unplugged, my cursor teleports to the top left corner on osu every time the driver restarts. My temporary solution to this was closing the driver, but I would like a feature where you can turn off the auto restart and just use the restart driver button when you want to use it! 

## Hardware/software information
Windows Version: 1803

Tablet Driver version: 0.2.3

Tablet model: CTH-480

YChvanov, 2019-08-06:
Cursor teleport is a known thing for a long time and since 0.2.1 auto restart can be disabled in bottom right corner

HilmTien, 2019-08-07:
Im blind. ahahha well thanks anyways!

closed 2019-08-07 by HilmTien