hawku/TabletDriver #462: Uninstalled Wacom Drivers but Hawku Drivers arent Working

xAbnn, 2019-08-16:
Just upgraded from Win 8.1 to Win 10, didn't have this problem before, I have two monitors as well, I can drag my cursor all the way to my other monitor, Tablet GUI does not change my area in a sense, its like the Wacom and tablet Gui are conflicting? idk

So, I've tried uninstalling and reinstalling Wacom drivers and Hwaku's many times but to no avail it still doesn't work. 

I have tried:
Disabling Windows Ink via Regedit.
Uninstalling OEM Wacom Drivers and Reinstalling latest Wacom drivers.
Looked at other Issues in the forums and tried but problem still persists.

Win 10 keeps installing Intous PT by itself every time I restart my computer.
Tablet GUI stops working if I remove Intous PT from devices. 

 
[image] If i restart my PC my tablet gets automatically detected and added to devices, but if i remove it Tablet GUI will stop working.

[image] I Uninstall all Wacom drivers as well (I ticked the remove software) restarted but Wacom Driver Downloader always shows up when I restart my PC and if i remove it im unable to use Tablet GUI.

Startup Log

[2019-08-16 21:44:30.205] \ Reading 'config/init.cfg'
[2019-08-16 21:44:30.205] >> Log startuplog.txt
[2019-08-16 21:44:30.206] Log file 'startuplog.txt' opened.
[2019-08-16 21:44:30.206] >> LogDirect true
[2019-08-16 21:44:30.206] Log direct print = True
[2019-08-16 21:44:30.206] >> Include "config\tablet.cfg"
[2019-08-16 21:44:30.206] \ Reading 'config\tablet.cfg'
[2019-08-16 21:44:30.206] >> #
[2019-08-16 21:44:30.206] >> # Example tablet definition:
[2019-08-16 21:44:30.206] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-08-16 21:44:30.206] >> #
[2019-08-16 21:44:30.206] >> # VID: 0x056a
[2019-08-16 21:44:30.206] >> # PID: 0x00dd
[2019-08-16 21:44:30.206] >> # HID Usage Page: 0x0D
[2019-08-16 21:44:30.206] >> # HID Usage: 0x01
[2019-08-16 21:44:30.206] >> #
[2019-08-16 21:44:30.206] >> #
[2019-08-16 21:44:30.206] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-08-16 21:44:30.206] >> #
[2019-08-16 21:44:30.206] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-08-16 21:44:30.207] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-16 21:44:30.208] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-08-16 21:44:30.208] >> ReportId 0x02
[2019-08-16 21:44:30.208] >> ReportLength 11
[2019-08-16 21:44:30.208] >> DetectMask 0x40
[2019-08-16 21:44:30.208] >> MaxX 14720
[2019-08-16 21:44:30.208] >> MaxY 9200
[2019-08-16 21:44:30.208] >> MaxPressure 1023
[2019-08-16 21:44:30.208] >> Width 147.2
[2019-08-16 21:44:30.208] >> Height 92.0
[2019-08-16 21:44:30.208] >> TabletFormat WacomDrivers
[2019-08-16 21:44:30.208] >> #
[2019-08-16 21:44:30.208] >> # Wacom CTL-470 
[2019-08-16 21:44:30.208] >> #
[2019-08-16 21:44:30.208] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-08-16 21:44:30.209] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-16 21:44:30.209] >> Name "Wacom CTL-470"
[2019-08-16 21:44:30.209] >> ReportId 0x02
[2019-08-16 21:44:30.209] >> ReportLength 10
[2019-08-16 21:44:30.209] >> DetectMask 0x40
[2019-08-16 21:44:30.209] >> MaxX 14720
[2019-08-16 21:44:30.209] >> MaxY 9200
[2019-08-16 21:44:30.209] >> MaxPressure 1023
[2019-08-16 21:44:30.209] >> Width 147.2
[2019-08-16 21:44:30.209] >> Height 92.0
[2019-08-16 21:44:30.209] >> InitFeature 0x02 0x02
[2019-08-16 21:44:30.209] >> #
[2019-08-16 21:44:30.209] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-08-16 21:44:30.209] >> #
[2019-08-16 21:44:30.209] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-08-16 21:44:30.211] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[... 359 lines cut ...]
[2019-08-16 21:44:30.238] 
[2019-08-16 21:44:30.238] Area:
[2019-08-16 21:44:30.238]   Desktop = 1920px x 1080px
[2019-08-16 21:44:30.238]   Screen Map = 1920px x 1080px @ X+0px, Y+0px
[2019-08-16 21:44:30.238]   Tablet =  77.20mm x 45.00mm @ Min(61.40mm, 52.50mm) Max(138.60mm, 97.50mm)
[2019-08-16 21:44:30.238]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-08-16 21:44:30.238] 
[2019-08-16 21:44:30.238] >> Start
[2019-08-16 21:44:30.240] TabletDriver started!
[2019-08-16 21:44:30.240] [STATUS] TABLET Wacom CTH-480
[2019-08-16 21:44:30.240] [STATUS] HID 056A 0302 FF0D 0001
[2019-08-16 21:44:30.240] [STATUS] WIDTH 152.00000
[2019-08-16 21:44:30.240] [STATUS] HEIGHT 95.00000
[2019-08-16 21:44:30.240] [STATUS] MAX_X 15200
[2019-08-16 21:44:30.240] [STATUS] MAX_Y 9500
[2019-08-16 21:44:30.240] [STATUS] MAX_PRESSURE 1023
[2019-08-16 21:44:30.240] [STATUS] AUX_BUTTONS 4
[2019-08-16 21:44:30.240] >> Log Off

## Hardware/software information
Windows Version: Latest 

Tablet Driver version: v0.23

Tablet model: CTH-480

xAbnn, 2019-08-16:
[image]
 So I restarted again and like I said it automatically starts to install new drivers?
[image]
Wacom Driver Downloader is there as well it always comes back after I Uninstall the wacom drivers

InfinityGhost, 2019-08-16:
Disable the wacom driver downloader

InfinityGhost, 2019-08-16:
Looking through the log, I saw this
`[2019-08-16 21:44:30.236] [ERROR] Can't open file 'config\wacom.cfg'`
Try redownloading and reinstalling TabletDriver.

xAbnn, 2019-08-17:
Disabled Wacom Driver Downloader, Program and features dont show any apps related to wacom.
I uninstalled the drivers via the remove_vmulti, deleted the entire Tablet Driver folder and restarted my PC.

Downloaded the Tablet Driver 0.23 and followed the instructions, configured my settings but still the problem persists. Restarted my PC and still same problem.

Startup Log.

[2019-08-17 12:41:56.423] Log file 'startuplog.txt' opened.
[2019-08-17 12:41:56.423] >> LogDirect true
[2019-08-17 12:41:56.423] Log direct print = True
[2019-08-17 12:41:56.423] >> Include "config\tablet.cfg"
[2019-08-17 12:41:56.427] \ Reading 'config\tablet.cfg'
[2019-08-17 12:41:56.427] >> #
[2019-08-17 12:41:56.427] >> # Example tablet definition:
[2019-08-17 12:41:56.427] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 12:41:56.427] >> #
[2019-08-17 12:41:56.427] >> # VID: 0x056a
[2019-08-17 12:41:56.427] >> # PID: 0x00dd
[2019-08-17 12:41:56.427] >> # HID Usage Page: 0x0D
[2019-08-17 12:41:56.427] >> # HID Usage: 0x01
[2019-08-17 12:41:56.427] >> #
[2019-08-17 12:41:56.427] >> #
[2019-08-17 12:41:56.427] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-08-17 12:41:56.427] >> #
[2019-08-17 12:41:56.427] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-08-17 12:41:56.428] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 12:41:56.428] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-08-17 12:41:56.428] >> ReportId 0x02
[2019-08-17 12:41:56.428] >> ReportLength 11
[2019-08-17 12:41:56.428] >> DetectMask 0x40
[2019-08-17 12:41:56.428] >> MaxX 14720
[2019-08-17 12:41:56.428] >> MaxY 9200
[2019-08-17 12:41:56.428] >> MaxPressure 1023
[2019-08-17 12:41:56.428] >> Width 147.2
[2019-08-17 12:41:56.428] >> Height 92.0
[2019-08-17 12:41:56.428] >> TabletFormat WacomDrivers
[2019-08-17 12:41:56.428] >> #
[2019-08-17 12:41:56.428] >> # Wacom CTL-470 
[2019-08-17 12:41:56.428] >> #
[2019-08-17 12:41:56.428] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 12:41:56.430] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-17 12:41:56.430] >> Name "Wacom CTL-470"
[2019-08-17 12:41:56.430] >> ReportId 0x02
[2019-08-17 12:41:56.430] >> ReportLength 10
[2019-08-17 12:41:56.430] >> DetectMask 0x40
[2019-08-17 12:41:56.430] >> MaxX 14720
[2019-08-17 12:41:56.430] >> MaxY 9200
[2019-08-17 12:41:56.430] >> MaxPressure 1023
[2019-08-17 12:41:56.430] >> Width 147.2
[2019-08-17 12:41:56.430] >> Height 92.0
[2019-08-17 12:41:56.430] >> InitFeature 0x02 0x02
[2019-08-17 12:41:56.430] >> #
[2019-08-17 12:41:56.430] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-08-17 12:41:56.430] >> #
[2019-08-17 12:41:56.430] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-08-17 12:41:56.431] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 12:41:56.431] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-08-17 12:41:56.431] >> ReportId 0x02
[2019-08-17 12:41:56.431] >> ReportLength 11
[2019-08-17 12:41:56.431] >> DetectMask 0x40
[2019-08-17 12:41:56.431] >> MaxX 14720
[2019-08-17 12:41:56.431] >> MaxY 9200
[2019-08-17 12:41:56.431] >> MaxPressure 1023
[2019-08-17 12:41:56.431] >> Width 147.2
[2019-08-17 12:41:56.431] >> Height 92.0
[2019-08-17 12:41:56.431] >> TabletFormat WacomDrivers
[2019-08-17 12:41:56.431] >> #
[2019-08-17 12:41:56.431] >> # Wacom CTH-470
[2019-08-17 12:41:56.431] >> #
[2019-08-17 12:41:56.431] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[... 352 lines cut ...]
[2019-08-17 12:41:56.464]   Report Length = 10 bytes
[2019-08-17 12:41:56.464]   Detect Mask = 0x40
[2019-08-17 12:41:56.464]   Ignore Mask = 0x00
[2019-08-17 12:41:56.464]   Pen button map = 'MOUSE1' '' '' 
[2019-08-17 12:41:56.464]   Aux button map = '' '' '' '' 
[2019-08-17 12:41:56.464]   Aux button count = 4
[2019-08-17 12:41:56.464]   Init feature report: { 0x02, 0x02 }
[2019-08-17 12:41:56.464] 
[2019-08-17 12:41:56.464] Area:
[2019-08-17 12:41:56.464]   Desktop = 1920px x 1080px
[2019-08-17 12:41:56.464]   Screen Map = 1920px x 1080px @ X+0px, Y+0px
[2019-08-17 12:41:56.464]   Tablet =  77.20mm x 45.00mm @ Min(61.40mm, 52.50mm) Max(138.60mm, 97.50mm)
[2019-08-17 12:41:56.464]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-08-17 12:41:56.464] 
[2019-08-17 12:41:56.464] >> Start
[2019-08-17 12:41:56.466] TabletDriver started!
[2019-08-17 12:41:56.466] [STATUS] TABLET Wacom CTH-480
[2019-08-17 12:41:56.466] [STATUS] HID 056A 0302 FF0D 0001
[2019-08-17 12:41:56.466] [STATUS] WIDTH 152.00000
[2019-08-17 12:41:56.466] [STATUS] HEIGHT 95.00000
[2019-08-17 12:41:56.466] [STATUS] MAX_X 15200
[2019-08-17 12:41:56.466] [STATUS] MAX_Y 9500
[2019-08-17 12:41:56.466] [STATUS] MAX_PRESSURE 1023
[2019-08-17 12:41:56.466] [STATUS] AUX_BUTTONS 4
[2019-08-17 12:41:56.466] >> Log Off

xAbnn, 2019-08-17:
My Device Manager also only shows one Pentablet HID is this correct?

xAbnn, 2019-08-17:
UPDATE: So I kind of fixed it? Downgraded to Tablet V0.2 and realized that Desktop Size was ticked on auto and my area worked (it did not go over to the other screen).

Reinstalled v0.23 and left that option at default (Ticked) and all is well now.

Setting the option to anything else makes the same problem. cant figure why. 

Thanks for the Help :D

closed 2019-08-17 by xAbnn