hawku/TabletDriver #463: Gaomon S620 Cursor not moving

messygamermode, 2019-08-17:
It says Connected to Gaomon S620 when I start the driver but my pen doesn't work at all.
Here is my startup log:
[2019-08-17 04:23:42.161] Log file 'startuplog.txt' opened.
[2019-08-17 04:23:42.161] >> LogDirect true
[2019-08-17 04:23:42.161] Log direct print = True
[2019-08-17 04:23:42.161] >> Include "config\tablet.cfg"
[2019-08-17 04:23:42.161] \ Reading 'config\tablet.cfg'
[2019-08-17 04:23:42.161] >> #
[2019-08-17 04:23:42.161] >> # Example tablet definition:
[2019-08-17 04:23:42.161] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 04:23:42.161] >> #
[2019-08-17 04:23:42.161] >> # VID: 0x056a
[2019-08-17 04:23:42.162] >> # PID: 0x00dd
[2019-08-17 04:23:42.162] >> # HID Usage Page: 0x0D
[2019-08-17 04:23:42.162] >> # HID Usage: 0x01
[2019-08-17 04:23:42.162] >> #
[2019-08-17 04:23:42.162] >> #
[2019-08-17 04:23:42.162] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-08-17 04:23:42.162] >> #
[2019-08-17 04:23:42.162] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-08-17 04:23:42.194] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 04:23:42.205] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-08-17 04:23:42.205] >> ReportId 0x02
[2019-08-17 04:23:42.205] >> ReportLength 11
[2019-08-17 04:23:42.205] >> DetectMask 0x40
[2019-08-17 04:23:42.205] >> MaxX 14720
[2019-08-17 04:23:42.205] >> MaxY 9200
[2019-08-17 04:23:42.205] >> MaxPressure 1023
[2019-08-17 04:23:42.205] >> Width 147.2
[2019-08-17 04:23:42.205] >> Height 92.0
[2019-08-17 04:23:42.205] >> TabletFormat WacomDrivers
[2019-08-17 04:23:42.205] >> #
[2019-08-17 04:23:42.205] >> # Wacom CTL-470 
[2019-08-17 04:23:42.205] >> #
[2019-08-17 04:23:42.205] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 04:23:42.237] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-17 04:23:42.248] >> Name "Wacom CTL-470"
[2019-08-17 04:23:42.248] >> ReportId 0x02
[2019-08-17 04:23:42.248] >> ReportLength 10
[2019-08-17 04:23:42.248] >> DetectMask 0x40
[2019-08-17 04:23:42.248] >> MaxX 14720
[2019-08-17 04:23:42.248] >> MaxY 9200
[2019-08-17 04:23:42.248] >> MaxPressure 1023
[2019-08-17 04:23:42.248] >> Width 147.2
[2019-08-17 04:23:42.248] >> Height 92.0
[2019-08-17 04:23:42.248] >> InitFeature 0x02 0x02
[2019-08-17 04:23:42.248] >> #
[2019-08-17 04:23:42.248] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-08-17 04:23:42.248] >> #
[2019-08-17 04:23:42.248] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-08-17 04:23:42.282] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 04:23:42.293] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-08-17 04:23:42.293] >> ReportId 0x02
[2019-08-17 04:23:42.293] >> ReportLength 11
[2019-08-17 04:23:42.293] >> DetectMask 0x40
[2019-08-17 04:23:42.293] >> MaxX 14720
[2019-08-17 04:23:42.293] >> MaxY 9200
[2019-08-17 04:23:42.293] >> MaxPressure 1023
[2019-08-17 04:23:42.293] >> Width 147.2
[2019-08-17 04:23:42.293] >> Height 92.0
[2019-08-17 04:23:42.293] >> TabletFormat WacomDrivers
[2019-08-17 04:23:42.293] >> #
[2019-08-17 04:23:42.293] >> # Wacom CTH-470
[2019-08-17 04:23:42.293] >> #
[2019-08-17 04:23:42.293] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2019-08-17 04:23:42.325] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-17 04:23:42.337] >> Name "Wacom CTH-470"
[2019-08-17 04:23:42.337] >> ReportId 0x02
[2019-08-17 04:23:42.337] >> ReportLength 10
[2019-08-17 04:23:42.337] >> DetectMask 0x40
[... 1069 lines cut ...]
[2019-08-17 04:23:44.959] Area:
[2019-08-17 04:23:44.959]   Desktop = 3280px x 1080px
[2019-08-17 04:23:44.959]   Screen Map = 3280px x 1080px @ X+1640px, Y+540px
[2019-08-17 04:23:44.959]   Tablet =  80.50mm x 55.40mm @ Min(114.52mm, 30.37mm) Max(195.02mm, 85.77mm)
[2019-08-17 04:23:44.959]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-08-17 04:23:44.959] 
[2019-08-17 04:23:44.959] >> Start
[2019-08-17 04:23:45.463] TabletDriver started!
[2019-08-17 04:23:45.463] [STATUS] TABLET Gaomon S620
[2019-08-17 04:23:45.463] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2019-08-17 04:23:45.463] [STATUS] WIDTH 165.10000
[2019-08-17 04:23:45.463] [STATUS] HEIGHT 101.60000
[2019-08-17 04:23:45.463] [STATUS] MAX_X 33020
[2019-08-17 04:23:45.463] [STATUS] MAX_Y 20320
[2019-08-17 04:23:45.463] [STATUS] MAX_PRESSURE 8191
[2019-08-17 04:23:45.463] [STATUS] AUX_BUTTONS 4
[2019-08-17 04:23:45.463] [STATUS] STARTED 1
[2019-08-17 04:23:45.463] >> Log Off

## Hardware/software information
Windows Version: 10

Tablet Driver version: 2.4pre

Tablet model: Gaomon S620

Mordans, 2019-08-29:
same problem here

RevENGE-code, 2019-09-02:
here is the solution