hawku/TabletDriver #464: Some solution for my desire.

mmnt1337, 2019-08-18:
I want the cursor to stop moving when I raised the pen, but I don’t have enough skill to come up with something or code myself

InfinityGhost, 2019-08-19:
That's not a bad idea at all. I'll figure out something on my fork eventually.

hawku, 2019-08-19:
@mmnt1337 For the relative mouse mode there is a feature like that.
Change mode to "Relative (Mouse)" and add `RelativeDragMove 1` to commands tab.
After that the cursor will only move if a pen pressure is detected.

@InfinityGhost 
https://github.com/hawku/TabletDriver/blob/d53eb54f16930ec76936295579f34043d8416f36/TabletDriverService/CommandHandler.Tablet.cpp#L725-L737

https://github.com/hawku/TabletDriver/blob/d53eb54f16930ec76936295579f34043d8416f36/TabletDriverService/OutputManager.cpp#L117-L121

mmnt1337, 2019-08-19:
> @mmnt1337 For the relative mouse mode there is a feature like that.
> Change mode to "Relative (Mouse)" and add `RelativeDragMove 1` to commands tab.
> After that the cursor will only move if a pen pressure is detected.

dosen't work on xp-pen g430 (v.0.2.3)
Cursor move when pen 2-3 cm away from tablet like it is would be before

mmnt1337, 2019-08-19:
[image]
[image]
Maybe something wrong?

hawku, 2019-08-19:
@mmnt1337 Sorry, I didn't remember that 0.2.3 doesn't have that feature.
Use InfinityGhost's version of TabletDriver: https://github.com/InfinityGhost/TabletDriver/releases/download/v0.2.4-pre/TabletDriverGUI.Release.zip

mmnt1337, 2019-08-19:
@hawku Oh,thank u very much

closed 2019-08-21 by mmnt1337