hawku/TabletDriver #465: Deco 01 V2 Tablet not found

s1ndy0, 2019-08-19:
Deco 01 V2 I bought today
[image]
PID seems to have changed?
What should i do?

## Hardware/software information
Windows Version: 
Windows10 1809
Tablet Driver version: 
0.2.3
Tablet model: 
Deco 01 V2

InfinityGhost, 2019-08-19:
Just for a test, edit the `tablet.cfg` and change the product ID yourself to see if the configuration still works.

s1ndy0, 2019-08-20:
Tablet init failed after changing product ID[(0x28BD 0x0902 0xFF0A 0x0001)→(0x28BD 0x0905 0xFF0A 0x0001)]
The official driver for xp-pen has never been installed.
Does xp-pen also need to uninstall the driver from Device Manager?
[image]
Try uninstalling them and you'll get back to Tablet not found
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3518406/startuplog.txt)

SyamuGame0, 2019-12-01:
hello. Did you solve this problem? I would like you to tell me if you know. I have the same problem now.