hawku/TabletDriver #466: Mouse issues

peturur, 2019-08-19:
When I install the driver through the ''install_vmulti_driver.bat'' my mouse always teleports to the far left of my 2nd monitor even when I don't touch it.

## Hardware/software information
Windows Version: 10

Tablet Driver version:  v0.1.5.5

Tablet model: VEIKK-S640

peturur, 2019-08-19:
Tells me in the console that the tablet is not found. I've unplugged and plugged my tablet into my computer but it still gives me the same results. I'm expecting that this is happening because it doesn't support veikk?, although hawku does.

InfinityGhost, 2019-08-23:
Make sure you followed the [installation instructions](https://github.com/hawku/TabletDriver#installation), it'll avoid a lot of issues like this.