hawku/TabletDriver #473: Gaomon S620 cursor does not move

Mordans, 2019-08-29:
The cursor is moving compatibility mode but not on standard game mode

> [2019-08-29 19:58:46.459] >> DetectMask 0xE0
> [2019-08-29 19:58:46.459] >> MaxX 44800
> [2019-08-29 19:58:46.459] >> MaxY 29600
> [2019-08-29 19:58:46.459] >> MaxPressure 8191
> [2019-08-29 19:58:46.459] >> Width 226.0
> [2019-08-29 19:58:46.459] >> Height 150.0
> [2019-08-29 19:58:46.459] >> #
> [2019-08-29 19:58:46.459] >> # XP-Pen G430S
> [2019-08-29 19:58:46.459] >> #
> [2019-08-29 19:58:46.459] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
> [2019-08-29 19:58:46.460] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0075 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
> [2019-08-29 19:58:46.471] >> CheckString Product "G430S"
> [2019-08-29 19:58:46.471] >> Name "XP-Pen G430S"
> [2019-08-29 19:58:46.471] >> ReportId 0x02
> [2019-08-29 19:58:46.471] >> ReportLength 8
> [2019-08-29 19:58:46.471] >> DetectMask 0x80
> [2019-08-29 19:58:46.471] >> MaxX 20320
> [2019-08-29 19:58:46.471] >> MaxY 15240
> [2019-08-29 19:58:46.471] >> MaxPressure 8191
> [2019-08-29 19:58:46.471] >> Width 101.6
> [2019-08-29 19:58:46.471] >> Height 76.2
> [2019-08-29 19:58:46.471] >> InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00
> [2019-08-29 19:58:46.471] >> #
> [2019-08-29 19:58:46.471] >> # XP-Pen G430
> [2019-08-29 19:58:46.471] >> #
> [2019-08-29 19:58:46.471] >> HIDTablet 0x28BD 0x0075 0xFF0A 0x0001
> [2019-08-29 19:58:46.472] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0075 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
> [2019-08-29 19:58:46.483] >> CheckString Product "TABLET G3"
> [2019-08-29 19:58:46.483] >> Name "XP-Pen G430"
> [2019-08-29 19:58:46.483] >> ReportId 0x02
> [2019-08-29 19:58:46.483] >> ReportLength 8
> [2019-08-29 19:58:46.483] >> DetectMask 0x80
> [2019-08-29 19:58:46.483] >> MaxX 45720
> [2019-08-29 19:58:46.483] >> MaxY 29210
> [2019-08-29 19:58:46.483] >> MaxPressure 2047
> [2019-08-29 19:58:46.483] >> Width 102.4
> [2019-08-29 19:58:46.483] >> Height 76.2
> [2019-08-29 19:58:46.483] >> InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00
> [2019-08-29 19:58:46.483] >> #
> [2019-08-29 19:58:46.483] >> # XP-Pen G540 Pro
> [2019-08-29 19:58:46.483] >> #
> [2019-08-29 19:58:46.483] >> HIDTablet 0x28BD 0x0061 0xFF0A 0x0001
> [2019-08-29 19:58:46.484] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0061 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
> [2019-08-29 19:58:46.495] >> Name "XP-Pen G540 Pro"
> [2019-08-29 19:58:46.495] >> ReportId 0x02
> [2019-08-29 19:58:46.495] >> ReportLength 8
> [2019-08-29 19:58:46.495] >> DetectMask 0x80
> [2019-08-29 19:58:46.495] >> MaxX 45720
> [2019-08-29 19:58:46.495] >> MaxY 29210
> [2019-08-29 19:58:46.495] >> MaxPressure 8191
> [2019-08-29 19:58:46.495] >> Width 147.0
> [2019-08-29 19:58:46.495] >> Height 101.6
> [2019-08-29 19:58:46.495] >> InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00
> [2019-08-29 19:58:46.495] >> #
> [2019-08-29 19:58:46.495] >> # XP-Pen G640
> [2019-08-29 19:58:46.495] >> #
> [2019-08-29 19:58:46.495] >> HIDTablet 0x28BD 0x0094 0xFF0A 0x0001
> [2019-08-29 19:58:46.496] [WARNING] Can't open HID tablet Vendor=0x28BD Product=0x0094 UsagePage=0xFF0A Usage=0x0001 Exclusive=False
> [2019-08-29 19:58:46.507] >> #CheckString Manufacturer "XP-PEN"
> [2019-08-29 19:58:46.507] >> #CheckString Product "STAR G640"
> [2019-08-29 19:58:46.507] >> Name "XP-Pen G640"
> [2019-08-29 19:58:46.507] >> ReportId 0x02
> [2019-08-29 19:58:46.507] >> ReportLength 8
> [2019-08-29 19:58:46.507] >> DetectMask 0x80
> [2019-08-29 19:58:46.507] >> MaxX 32000
> [2019-08-29 19:58:46.507] >> MaxY 20000
> [2019-08-29 19:58:46.507] >> MaxPressure 8191
> [2019-08-29 19:58:46.507] >> Width 160.0
[... 92 lines cut ...]
> [2019-08-29 19:59:16.586] >> MaxX 24384
> [2019-08-29 19:59:16.586] >> MaxY 15240
> [2019-08-29 19:59:16.586] >> MaxPressure 4095
> [2019-08-29 19:59:16.586] >> Width 121.92
> [2019-08-29 19:59:16.586] >> Height 76.20
> [2019-08-29 19:59:16.586] >> InitStrings 100 200
> [2019-08-29 19:59:16.586] >> AuxReportId 0x08
> [2019-08-29 19:59:16.586] >> AuxReportLength 12
> [2019-08-29 19:59:16.586] >> AuxDetectMask 0xE0
> [2019-08-29 19:59:16.586] >> AuxgnoreMask 0x80
> [2019-08-29 19:59:16.586] >> AuxCustomData ReportId Source=0
> [2019-08-29 19:59:16.586] >> AuxCustomData ButtonsLow Source=4 SourceMask=0x0F
> [2019-08-29 19:59:16.586] >> AuxCustomData Detect Source=1
> [2019-08-29 19:59:16.586] >> AuxButtonCount 4
> [2019-08-29 19:59:16.586] >> #
> [2019-08-29 19:59:16.586] >> # Huion H640P
> [2019-08-29 19:59:16.586] >> #
> [2019-08-29 19:59:16.586] >> USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"

## Hardware/software information
Windows Version: windows 10

Tablet Driver version: v0.3.0-pre

Tablet model: Gaomon S620

Mordans, 2019-08-29:
I install the driver Huion tablet driver

RevENGE-code, 2019-09-02:
here is the solution https://www.reddit.com/r/osugame/comments/cywlbt/how_to_install_hawku_driver_for_the_gaomon_s620/