hawku/TabletDriver #476: Won't detect my G430S

chubssu, 2019-08-31:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: XP-PEN G430S

[2019-08-31 21:17:00.182] >> DetectMask 0x40
[2019-08-31 21:17:00.182] >> MaxX 21600
[2019-08-31 21:17:00.182] >> MaxY 13500
[2019-08-31 21:17:00.182] >> MaxPressure 1023
[2019-08-31 21:17:00.182] >> Width 216.000
[2019-08-31 21:17:00.182] >> Height 135.000
[2019-08-31 21:17:00.182] >> InitFeature 0x02 0x02
[2019-08-31 21:17:00.182] >> # Wacom CTH-460
[2019-08-31 21:17:00.182] >> HIDTablet 0x056A 0x00D1 0xFF0D 0x0001
[2019-08-31 21:17:00.184] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.184] >> HIDTablet 0x056A 0x00D1 0x000D 0x0001
[2019-08-31 21:17:00.185] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.185] >> Name "Wacom CTH-460"
[2019-08-31 21:17:00.185] >> ReportLength 9
[2019-08-31 21:17:00.185] >> DetectMask 0x40
[2019-08-31 21:17:00.185] >> MaxX 14720
[2019-08-31 21:17:00.185] >> MaxY 9200
[2019-08-31 21:17:00.186] >> MaxPressure 1023
[2019-08-31 21:17:00.186] >> Width 147.200
[2019-08-31 21:17:00.186] >> Height 92.000
[2019-08-31 21:17:00.186] >> InitFeature 0x02 0x02
[2019-08-31 21:17:00.186] >> # Wacom CTH-460(A)
[2019-08-31 21:17:00.186] >> HIDTablet 0x056A 0x00D6 0xFF0D 0x0001
[2019-08-31 21:17:00.187] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.187] >> HIDTablet 0x056A 0x00D6 0x000D 0x0001
[2019-08-31 21:17:00.189] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.189] >> Name "Wacom CTH-460(A)"
[2019-08-31 21:17:00.189] >> ReportLength 9
[2019-08-31 21:17:00.189] >> DetectMask 0x40
[2019-08-31 21:17:00.189] >> MaxX 14720
[2019-08-31 21:17:00.189] >> MaxY 9200
[2019-08-31 21:17:00.189] >> MaxPressure 1023
[2019-08-31 21:17:00.189] >> Width 147.200
[2019-08-31 21:17:00.189] >> Height 92.000
[2019-08-31 21:17:00.189] >> InitFeature 0x02 0x02
[2019-08-31 21:17:00.189] >> # Wacom CTH-461
[2019-08-31 21:17:00.189] >> HIDTablet 0x056A 0x00D2 0xFF0D 0x0001
[2019-08-31 21:17:00.191] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.191] >> HIDTablet 0x056A 0x00D2 0x000D 0x0001
[2019-08-31 21:17:00.192] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.193] >> Name "Wacom CTH-461"
[2019-08-31 21:17:00.193] >> ReportLength 9
[2019-08-31 21:17:00.193] >> DetectMask 0x40
[2019-08-31 21:17:00.193] >> MaxX 14720
[2019-08-31 21:17:00.193] >> MaxY 9200
[2019-08-31 21:17:00.193] >> MaxPressure 1023
[2019-08-31 21:17:00.193] >> Width 147.200
[2019-08-31 21:17:00.193] >> Height 92.000
[2019-08-31 21:17:00.193] >> InitFeature 0x02 0x02
[2019-08-31 21:17:00.193] >> # Wacom CTH-461(A)
[2019-08-31 21:17:00.193] >> HIDTablet 0x056A 0x00D7 0xFF0D 0x0001
[2019-08-31 21:17:00.195] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.195] >> HIDTablet 0x056A 0x00D7 0x000D 0x0001
[2019-08-31 21:17:00.196] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-31 21:17:00.196] >> Name "Wacom CTH-461(A)"
[2019-08-31 21:17:00.196] >> ReportLength 9
[2019-08-31 21:17:00.196] >> DetectMask 0x40
[2019-08-31 21:17:00.196] >> MaxX 14720
[2019-08-31 21:17:00.196] >> MaxY 9200
[2019-08-31 21:17:00.196] >> MaxPressure 1023
[2019-08-31 21:17:00.196] >> Width 147.200
[2019-08-31 21:17:00.196] >> Height 92.000
[2019-08-31 21:17:00.196] >> InitFeature 0x02 0x02
[... 214 lines cut ...]
[2019-08-31 21:17:00.250] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2019-08-31 21:17:00.250] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2019-08-31 21:17:00.250] 
[2019-08-31 21:17:00.250]   Driver version: 0.2.3
[2019-08-31 21:17:00.250]   Windows version: Microsoft Windows NT 6.2.9200.0
[2019-08-31 21:17:00.250]   Windows product: Windows 10 Enterprise
[2019-08-31 21:17:00.250]   Windows release: 1809
[2019-08-31 21:17:00.250] 
[2019-08-31 21:17:00.250] >> CheckTablet
[2019-08-31 21:17:00.251] [ERROR] Tablet not found!
[2019-08-31 21:17:00.251] [ERROR] Check the list of supported tablets from the GitHub page.
[2019-08-31 21:17:00.251] [ERROR] http://github.com/hawku/TabletDriver
Driver stopped!

Any help please? I'm pretty dumb when it comes to these kinds of stuff

InfinityGhost, 2019-09-01:
Send the `startuplog.txt` file as an attachment, it'll give us more information on what's causing it to not be detected.

Cendri11oN, 2019-09-13:
I also have this problem I dont know why,,,
Here's my txt file
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3610220/startuplog.txt)

riley-badour, 2019-09-13:
can you please Send  the `startuplog.txt`  with the debug checkboxed checked