hawku/TabletDriver #486: Petition to configure hawku for cintiq tablets

fammerino, 2019-09-14:
I was wondering if there will ever be possible to use a cintiq tablet with this driver, i see that its probably hard since its a screen based tablet, but i play without the screen so its basically just a more expensive tablet with much better tracking. but mapping an area is really hard since it doesnt have the same sort of positional numbers as normal wacom tablets.

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model: cintiq 13hd

InfinityGhost, 2019-09-15:
It's very likely to be possible. What's the difference between mapping areas on cintiq and normal wacom tablets?

fammerino, 2019-10-07:
Well the main difference is that on the cintiq you have a display, so you
calibrate the area from clicking points on the tablet display, but by only
using the power cable and a USB for the cintiq it acts as a normal tablet,
there is no interface for mapping the area, you simply have to do it by
eyesight, or physically cut out the area in a piece of paper and use that
(like I did)