hawku/TabletDriver #488: Tablet not found! Problem with G430s

UberAtziri, 2019-09-18:
Device works with xp-pen driver but TabletDriver cant detect it. I tried to unninstal official driver, it didn't help. Also tried to find wacom drivers in device manager, nothing there with "Wacom". Please Help!

startuplog file included.

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3626512/startuplog.txt)

Tablet model: G430s

UberAtziri, 2019-09-18:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3628394/startuplog.txt)
Log with debug checkbox.

Kyoukai304, 2019-10-08:
I have the exact same problem and tablet as you. I hope there is a fix soon.