hawku/TabletDriver #489: Code refactor when instantiate MainWindow.xaml

quangaming2929, 2019-09-19:
I think this is a minor improvement but start an Window in App contructor is not a good place because the App class doesn't have time to Load it's resource. I think you should consider move it to Application.Startup event or set the StartUri to MainWindow.xaml

InfinityGhost, 2019-09-22:
IMO the entire GUI needs to be rebuilt from the ground up. There's a lot of unmanageable code and repeated code within the entire GUI project.

hawku, 2019-09-22:
@InfinityGhost Yeah, that is true.
The GUI wasn't from start designed for the current feature set and I have just kept piling up things on top of that old code.
Rebuilding the code would also mean that a support for multiple tablets and profiles could be added, but currently I don't have time for that.

InfinityGhost, 2019-09-22:
I would like to make the entire project cross platform as well, but I don't even know where to start since the project has become so expansive now.

quangaming2929, 2019-09-26:
@InfinityGhost I think cross platform for driver is a hard thing to do because not only you can't use WPF to make GUI but also you need to write library for all tablet models for each platforms

quangaming2929, 2019-09-26:
I think one way to make it easier to read is extract each functionality to a new UserControls. Sometimes partial class make it harder to analyze code if you don't use IDE Find all references feature

InfinityGhost, 2019-09-26:
> @InfinityGhost I think cross platform for driver is a hard thing to do because not only you can't use WPF to make GUI but also you need to write library for all tablet models for each platforms

There are open source multiplatform GUI solutions that would solve that issue

> I think one way to make it easier to read is extract each functionality to a new UserControls. Sometimes partial class make it harder to analyze code if you don't use IDE Find all references feature

The design of the MainWindow for sure has way too many individual components in a single file that would definitely take some time to split apart into separate controls. I think a quick start there would be to split at least the duplicate area canvases into a single "AreaSetter" control