hawku/TabletDriver #496: Smoothing Filter makes Tablet unusable

Knallli, 2019-09-26:
When I enable cursor smoothing  I only get 64-68packets/s report rate and incredible amounts of delay/lag even at low ms.
Polling rate is fine when I disable smoothing.

I've tried everything already (Different USB Ports, reinstalling driver, reinstalling wacom, uninstalling wacom driver, different driver versions, reinstallation of vmulti driver, new config file)

## Hardware/software information
Windows Version: Win 10 1903

Tablet Driver version: latest

Tablet model: CTL-480

Knallli, 2019-09-26:
noticed that this is propably a duplicate of https://github.com/hawku/TabletDriver/issues/440

jjhhgg100123, 2019-09-26:
Try [Infinity Ghost's fork](https://github.com/InfinityGhost/TabletDriver) instead.