hawku/TabletDriver #497: Problem with TabletDriverService.exe

Cheetek, 2019-09-26:
I have a problem with turning on the last TabletDriver because I recently wanted to play osu!, And here suddenly he writes me something like this:
[image]

 I don't know what it is, but when I tried to turn it off by Task Manager, he writes to me: [image] Please help.

Cheetek, 2019-09-26:
Oh, and Hardware/software information
Windows Version: 10 (home)

Tablet Driver version: Latest 

Tablet model: XP-Pen G430S

jjhhgg100123, 2019-09-26:
Use [Infinity ghost's](https://github.com/InfinityGhost/TabletDriver) fork instead.

Cheetek, 2019-09-27:
Yes i tried kill that process tab but I see text: The operation could not be completed. Acces is denided.

When i downloaded that Infinity ghost's fork instead, them I have that error: 
[image]

Cheetek, 2019-09-27:
And when I tried close this app with command propot, them I have that: 
[image]

jjhhgg100123, 2019-09-28:
Did you right click and run the command prompt as an admin?

Try deleting the folder normally, or with the command prompt.

Cheetek, 2019-09-28:
Ye I tried delete a folder, but that proccess is nonstop active. When I opened command prompt as an admin, and when I type that command, then I saw that: 
[image]

Cheetek, 2019-09-28:
And when I type that command I saw that: 
[image]

Cheetek, 2019-09-28:
[image]

jjhhgg100123, 2019-09-28:
Boot in to safe mode and delete the folder.

InfinityGhost, 2019-09-29:
There's some permission issue here for sure, I would definitely follow this:

> Boot in to safe mode and delete the folder.

InfinityGhost, 2019-09-29:
Something is starting the process without your permission, which is likely corrupted and/or has been compromised

Cheetek, 2019-09-29:
Ye, when I start my pc, them I saw that: 
[image]

InfinityGhost, 2019-09-29:
Have you deleted that executable in safe mode?

Cheetek, 2019-09-29:
But how can I delete him in safe mode?

InfinityGhost, 2019-09-29:
Boot into safe mode and all startup applications will be disabled no matter what. This will allow you to delete the file itself and it won't open on startup anymore

Cheetek, 2019-09-29:
Ok, but how can I boot into safe mode?

InfinityGhost, 2019-09-29:
[image]
Open settings, Update and Security, Recovery

Cheetek, 2019-09-29:
Ok, and when I boot my pc to safe mode, then what?

InfinityGhost, 2019-09-29:
Delete the tabletdriverservice.exe that was running

Cheetek, 2019-09-29:
Ok i deleted. What now?

Cheetek, 2019-09-29:
Omg that worked, but now I'm need that TabletDriverService: 
[image]

Cheetek, 2019-09-29:
Ok, i'm reinstalled that driver, but now I have that error: 
[image]

Cheetek, 2019-09-29:
Ok i'm started the driver with administrator. Im closing that.

closed 2019-09-29 by Cheetek