hawku/TabletDriver #504: driver stopped. restarting!! check console!!

rubberdoe, 2019-10-02:
this started happening 4 days ago and I'm not sure whats wrong, I tried using it on different computers and it still didn't work. also gives me "[ERROR] Tablet not found!" even though its connected. please help

## Hardware/software information
Windows Version: 10

Tablet Driver version: TabletDriverV0.1.5.5.Devocub.Edition

Tablet model: CTH-480

InfinityGhost, 2019-10-03:
Try a newer version of TabletDriver, and see if the problem persists.

rubberdoe, 2019-10-06:
> Try a newer version of TabletDriver, and see if the problem persists.

already tried this, did not work sadly.

InfinityGhost, 2019-10-07:
The only reason I could see this happening is that the `config` directory doesn't contain `wacom.cfg` with the CTH-480 configuration.

rubberdoe, 2019-10-08:
alright so yesterday i figured out what the problem was. I just needed a new wire LOL, thanks for the help though :D

closed 2019-10-08 by rubberdoe