hawku/TabletDriver #506: Tablet not found XP-PEN G430S

Kyoukai304, 2019-10-08:
Good day! I can't seem to make the drivers to work. It always says that the tablet not found. I already uninstalled previous drivers and tried reinstalling the hawku drivers (vmulti driver) to no avail. I also tried to restart my pc multiple times. I tried searching for solutions online and nothing seems to work. I am using the latest version of the drivers. I also already have dotnet framework and vc_redist. I honestly don't know what causes this.

## Hardware/software information
Windows Version: 

Tablet Driver version: TabletDriverV0.2.3

Tablet model: XP-PEN G430S
[image]

maxxximo0, 2019-10-08:
I have the same problem with the same tablet..

Marblelemons, 2019-10-09:
Have the same problem with the G640. Noticeably, the software keeps thinking my tablet is another tablet (currently it thinks it's a Wacom PTH-850, but usually it thinks it's a Wacom CTL-4100 Bluetooth).

rma2002, 2019-10-12:
Same! I have G640 though

haboutnnah, 2019-12-23:
Same
```
[2019-12-23 18:58:27.733] \ Reading 'config/init.cfg'
[2019-12-23 18:58:27.733] >> Log startuplog.txt
[2019-12-23 18:58:27.733] Log file 'startuplog.txt' opened.
[2019-12-23 18:58:27.733] >> LogDirect true
[2019-12-23 18:58:27.734] Log direct print = True
[2019-12-23 18:58:27.734] >> Include "config\tablet.cfg"
[2019-12-23 18:58:27.734] \ Reading 'config\tablet.cfg'
[2019-12-23 18:58:27.734] >> #
[2019-12-23 18:58:27.734] >> # Example tablet definition:
[2019-12-23 18:58:27.734] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-12-23 18:58:27.734] >> #
[2019-12-23 18:58:27.734] >> # VID: 0x056a
[2019-12-23 18:58:27.734] >> # PID: 0x00dd
[2019-12-23 18:58:27.734] >> # HID Usage Page: 0x0D
[2019-12-23 18:58:27.734] >> # HID Usage: 0x01
[2019-12-23 18:58:27.734] >> #
[2019-12-23 18:58:27.734] >> #
[2019-12-23 18:58:27.734] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-12-23 18:58:27.734] >> #
[2019-12-23 18:58:27.734] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-12-23 18:58:27.735] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-12-23 18:58:27.735] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-12-23 18:58:27.735] >> ReportId 0x02
[2019-12-23 18:58:27.735] >> ReportLength 11
[2019-12-23 18:58:27.735] >> DetectMask 0x40
[2019-12-23 18:58:27.735] >> MaxX 14720
[2019-12-23 18:58:27.735] >> MaxY 9200
[2019-12-23 18:58:27.735] >> MaxPressure 1023
[2019-12-23 18:58:27.735] >> Width 147.2
[2019-12-23 18:58:27.735] >> Height 92.0
[2019-12-23 18:58:27.735] >> TabletFormat WacomDrivers
[2019-12-23 18:58:27.735] >> #
[2019-12-23 18:58:27.735] >> # Wacom CTL-470 
[2019-12-23 18:58:27.735] >> #
[2019-12-23 18:58:27.735] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-12-23 18:58:27.736] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-12-23 18:58:27.736] >> Name "Wacom CTL-470"
[2019-12-23 18:58:27.736] >> ReportId 0x02
[2019-12-23 18:58:27.736] >> ReportLength 10
[2019-12-23 18:58:27.736] >> DetectMask 0x40
[2019-12-23 18:58:27.736] >> MaxX 14720
[2019-12-23 18:58:27.736] >> MaxY 9200
[2019-12-23 18:58:27.736] >> MaxPressure 1023
[2019-12-23 18:58:27.736] >> Width 147.2
[2019-12-23 18:58:27.736] >> Height 92.0
[2019-12-23 18:58:27.736] >> InitFeature 0x02 0x02
[2019-12-23 18:58:27.736] >> #
[2019-12-23 18:58:27.736] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-12-23 18:58:27.736] >> #
[2019-12-23 18:58:27.736] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-12-23 18:58:27.737] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-12-23 18:58:27.737] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-12-23 18:58:27.737] >> ReportId 0x02
[2019-12-23 18:58:27.737] >> ReportLength 11
[2019-12-23 18:58:27.737] >> DetectMask 0x40
[2019-12-23 18:58:27.737] >> MaxX 14720
[2019-12-23 18:58:27.737] >> MaxY 9200
[2019-12-23 18:58:27.737] >> MaxPressure 1023
[2019-12-23 18:58:27.737] >> Width 147.2
[2019-12-23 18:58:27.737] >> Height 92.0
[2019-12-23 18:58:27.737] >> TabletFormat WacomDrivers
[2019-12-23 18:58:27.737] >> #
[2019-12-23 18:58:27.737] >> # Wacom CTH-470
[2019-12-23 18:58:27.737] >> #
[2019-12-23 18:58:27.737] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2019-12-23 18:58:27.738] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-12-23 18:58:27.738] >> Name "Wacom CTH-470"
[2019-12-23 18:58:27.738] >> ReportId 0x02
[... 1547 lines cut ...]
[2019-12-23 18:58:27.873] [STATUS] COMMANDS AntiSmoothingFilter Anti AntiSmoothing AuxButtonCount AuxDetectMask AuxHID AuxIgnoreMask 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS AuxReportId AuxReportLength Benchmark Bench ButtonMap AuxButtonMap AuxButtons Buttons 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2019-12-23 18:58:27.873] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2019-12-23 18:58:27.873] 
[... cut ...] on: 0.2.3
[2019-12-23 18:58:27.873]   Windows version: Microsoft Windows NT 6.2.9200.0
[2019-12-23 18:58:27.873]   Windows product: Windows 10 Enterprise LTSC 2019
[2019-12-23 18:58:27.873]   Windows release: 1809
[2019-12-23 18:58:27.873] 
[2019-12-23 18:58:27.873] >> CheckTablet
[2019-12-23 18:58:27.873] [ERROR] Tablet not found!
[2019-12-23 18:58:27.873] [ERROR] Check the list of supported tablets from the GitHub page.
[2019-12-23 18:58:27.873] [ERROR] http://github.com/hawku/TabletDriver
```

matthewleemle, 2020-06-26:
same problem, hopefully gets solved soon