hawku/TabletDriver #514: Stroke malfunctioning

selshas, 2019-10-24:
[image]

[image]

[image]

hi! As an Graphic artist, I've newly found out this driver, to increase my production efficiency, because Wacom official driver has various problems that make me annoying.

So, i was very satisfied with this driver's performance, but there is a critical issue, so that i should not take to use this driver for my artwork.

image 0 and 1 shows some separation of strokes but actually these strokes are not separated by my hand. my hand moved continually but somehow, the driver stopped to drawing lines for a moment. as you see, it is not from drawing software.

and look at image 2. I let idle my stylus pen on tablet, just not touched on draw surface.
and then, somehow my windows startup menu popped up. so i concluded this malfunction originated from miss positioned cursor and button inputs. somehow, while the pen is on the draw surface(while the main body is detecting pen) the cursor has been teleported to left bottom corner of the screen and mouse click occurred in same time and keep this denial state while few miliseconds and refreshing its position and pressure by next inputs after resurrect input.

it seems occurring periodically. might have some deadlock status on threads.

## Hardware/software information
Windows Version:  Win10

Tablet Driver version: 0.2.3

Tablet model: PTH-651

selshas, 2019-10-24:
Thanks to you reply.

I've tried to changing DetectMask to 0xE0 in the section of PTH-651 on Wacom.cfg, and Unfortunately, this solution does not work to fix the problem.

so I've recorded some debug logs reproduce the bug. 
To help exact locating of issue and conditions, i've attached three debug logs and startup log.

[startuplog.txt](https://github.com/hawku/TabletDriver/files/3766386/startuplog.txt)
[[debugLog3]debug_2019-10-24_17_12_36.txt](https://github.com/hawku/TabletDriver/files/3766385/debugLog3.debug_2019-10-24_17_12_36.txt)
[[debugLog1]debug_2019-10-24_17_10_25.txt](https://github.com/hawku/TabletDriver/files/3766387/debugLog1.debug_2019-10-24_17_10_25.txt)
[[debugLog2]debug_2019-10-24_17_10_52.txt](https://github.com/hawku/TabletDriver/files/3766388/debugLog2.debug_2019-10-24_17_10_52.txt)

ishanpm, 2020-03-24:
I solved this issue on my Wacom CTL-471 by adding `ReportId 0x02` to the entry in tablet.cfg. That line is missing from some of the Wacom models for some reason.