hawku/TabletDriver #516: XP-Pen G430S Not Detected

DeadpixeI, 2019-10-28:
Tablet isn't being recognized by the drivers, anyone else having this issue? Tablet is plugged in and is recognized by Windows. Not sure what the issue is.
Start Up Log is below:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3776910/startuplog.txt)

## Hardware/software information
Windows Version: Windows 10 Pro Version 1903

Tablet Driver version: TabletDriverV0.2.3

Tablet model: XP-Pen G430S

Marblelemons, 2019-10-29:
Seems this has been an issue for a lot of people; Gaomon and XP-Pen tablets have been hit in particular.
It seems like spamming restart will _eventually_ work, and it looks like it's just total RNG.

Hawku fix when.

Marblelemons, 2019-10-29:
A few more things I should probably note

I'm pretty sure this bug is triggered by area changes. I did get it to consistently work with my xp-pen g640 at some point, however after adjusting my tablet area again I received the bug once more. So far I haven't been able to recreate my fix. All I remember is uninstalling and reinstalling my drivers over and over.

Sometimes I'll get a cursor that oscillates back and forth between 2 points on my screen - one point on my newly changed tablet area, and the other on the area I had when I started up the drivers **for the first time after starting my computer.**

@hawku fix when.

xCracked, 2019-12-09:
Check [this](https://www.reddit.com/r/osugame/comments/dnwtdk/potential_fix_for_xppen_g430s_and_community/?utm_medium=android_app&utm_source=share). It worked for me.