hawku/TabletDriver #521: CTL-480 [ERROR] Tablet not found!

Alphaheat, 2019-11-01:
Always running into the error saying that my tablet is not found. I've removed all Wacom drivers from Device Manager and looked into control panel's "Uninstall a program" for anything related to Wacom (couldn't find anything relating to any tablets let alone Wacom). I also re-downloaded the Wacom driver for my specific tablet to remove it again and the problem persists. I have removed the vmulti driver and reinstalled it every step that I mentioned above as well. 

## Hardware/software information
Windows Version: Windows 10 Ver. 10.0.18362 Build 18362

Tablet Driver version: 0.2.3

Tablet model: CTL-480

Alphaheat, 2019-11-01:
Found a fix for it. The .cfg files in the config folder were busted for some reason. Re-downloaded the TabletDriver zip folder again and ran it wthout any issues. Problem has now been solved.

closed 2019-11-01 by Alphaheat

asd694427704, 2020-02-27:
I have the same problem as you, can you teach me how to solve it?