hawku/TabletDriver #527: Cannot install Vmulti

Elaborateshoe, 2019-11-10:
[image]

I cannot install Vmulti even though I have run it as an admin, removed vmulti as admin, and made sure all Wacom drivers are not installed. Any help is appreciated

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: Wacom CTH-470

allbombson, 2019-11-11:
I see u have a Bamboo CTH-470, same as me. What version of Wacom drivers did u have installed before?

Elaborateshoe, 2019-11-11:
>I have 
Driver 5.3.5-3 (Windows XP, Vista, Win 7, 8, & 10)

38.2 MB 8/20/2015

Elaborateshoe, 2019-11-11:
> 
> 
> I see u have a Bamboo CTH-470, same as me. What version of Wacom drivers did u have installed before?

I have Driver 5.3.5-3 (Windows XP, Vista, Win 7, 8, & 10)

38.2 MB 8/20/2015

closed 2019-11-11 by Elaborateshoe