hawku/TabletDriver #531: Problem with using 2 Tablets

BattleBiscuit97, 2019-11-15:
So I've got myself a xppen 16pro with fancy screen and all.
Before I was using my CTH-480 with this driver for osu.

It stopped working after installing the xppen drivers and software.

Is there a way for me to use both tablets? 
(The xppen for drawing obviously, cth 480 with this driver for osu!)

I've gotten a Bluescreen after unplugging the xppen while running TabletDriverGUI to test if the cth works with the xppen drivers still installed.

I deinstalled the xppen driver/software ,so i can play osu, but the xppen isn't working because of that...

Anyone having a similar setup?
 
## Hardware/software information
Windows Version:  10 64Bit

Tablet Driver version:  TDG 0.2.3
                                  XP-Pen : 20190903

Tablet model:  CTH-480 , XP-Pen Artist 16 Pro

SBAPKat, 2019-11-16:
Hi, just to confirm, once you uninstall the xp pen driver, does hawku's work normally?
If that's the case, it should just be a matter of adding support for the XP pen.

BattleBiscuit97, 2019-11-17:
@SBAPKat 
Well yes, but I need the xppen driver to work aswell since I need the xppen tablet for drawing.
So I was thinking about having something that stops the xppen driver stuff and then I start TDG?

SBAPKat, 2019-11-17:
You can definitely draw with hawku's on XP Pen, you just need to activate Windows ink mode to get pressure levels back. Only bummer is that shortcut buttons on the tablet are not functioning.

BattleBiscuit97, 2019-11-17:
@SBAPKat 
That's what I was trying to prevent. I need the shortcuts. 
Is there a way to run both drivers? Like disabling the services for xppen and then starting TDG?

SBAPKat, 2019-11-17:
You should be able to kill the XP pen driver in task manager, once you've removed the tablet, and then plugging the Wacom, starting TDG once that's done.

BattleBiscuit97, 2019-11-18:
@SBAPKat 
Alright. So:
It works like I imagined , BUT the absolute mode in TDG doesn't work. I have to choose the SendInput (debug) mode.

Is there any drawbacks to using the SendInput over the absolute mode?

thanks for the help up till now^^

SBAPKat, 2019-11-21:
I haven't noticed any drawback to using the debug method, you could run into issues with this mode, as I don't know much about it