hawku/TabletDriver #533: Tablet not working when driver is running

tillann, 2019-11-16:
When I run the driver, my tablet isn't registering any input at all. It works fine when drivers aren't on but as soon as i start it it just wont move my cursor. 

Main page screenshot: https://i.imgur.com/yVMxLYp.png
Console screenshot: https://i.imgur.com/5ohGWwP.png

I don't have any other drivers installed that I'm aware of.

## Hardware/software information
Windows Version: Latest

Tablet Driver version: V0.2.3

Tablet model: XP Pen G640

tillann, 2019-11-16:
Forgot to select an option for ''Set Area'' in the screenshot, but nothing changes depending on what I choose there anyway.

SBAPKat, 2019-11-16:
Hi,
There are multiple versions of this tablet.
Please go to the console tab and tick the Debug box before restarting the driver with the appropriate button. Once that's done post the contents of startuplog.txt her so that someone may help you.

tillann, 2019-11-16:
https://i.imgur.com/90adQod.png 

So apparently it is registering, but my cursor still isn't moving? I'm very confused

tillann, 2019-11-16:
Here's the startuplog contents

[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice] 
[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice] HID Device: Vendor: 'XP-PEN' Product: 'STAR G640', Serial: ''
[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0094
[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice]   Usage Page: 0x000D, Usage: 0x000E
[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice]   FeatureLen: 3, InputLen: 0, OutputLen: 0
[2019-11-16 22:57:04.692] [DEBUG] [HIDDevice] 
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice] HID Device: Vendor: 'XP-PEN' Product: 'STAR G640', Serial: ''
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0094
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   Usage Page: 0x000C, Usage: 0x0036
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 5, OutputLen: 0
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice] 
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice] HID Device: Vendor: 'XP-PEN' Product: 'STAR G640', Serial: ''
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0094
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   Usage Page: 0xFF0A, Usage: 0x0001
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 8, OutputLen: 8
[2019-11-16 22:57:04.693] [DEBUG] [HIDDevice] 
[2019-11-16 22:57:04.693] >> GetCommands
[2019-11-16 22:57:04.693] [STATUS] COMMANDS AntiSmoothingFilter Anti AntiSmoothing AuxButtonCount AuxDetectMask AuxHID AuxIgnoreMask 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS AuxReportId AuxReportLength Benchmark Bench ButtonMap AuxButtonMap AuxButtons Buttons 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2019-11-16 22:57:04.693] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2019-11-16 22:57:04.693] 
[2019-11-16 22:57:04.693]   Driver version: 0.2.3
[2019-11-16 22:57:04.693]   Windows version: Microsoft Windows NT 6.2.9200.0
[2019-11-16 22:57:04.693]   Windows product: Windows 10 Home
[2019-11-16 22:57:04.694]   Windows release: 1903
[2019-11-16 22:57:04.694] 
[2019-11-16 22:57:04.694] >> CheckTablet
[2019-11-16 22:57:04.694] >> DesktopSize 3840 1088
[2019-11-16 22:57:04.694] Desktop size = (3840.00 px x 1088.00 px)
[2019-11-16 22:57:04.694] >> ScreenArea 1920 1080 0 0
[2019-11-16 22:57:04.694] Screen area = (w=1920.00, h=1080.00, x=0.00, y=0.00)
[2019-11-16 22:57:04.694] >> TabletArea 60 46 56.32 43.83
[2019-11-16 22:57:04.694] Tablet area: (60.00 mm x 46.00 mm X+56.32 mm Y+43.83 mm)
[2019-11-16 22:57:04.694] >> Rotate 0
[2019-11-16 22:57:04.694] Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-11-16 22:57:04.694] >> OutputMode Absolute
[2019-11-16 22:57:04.694] Output Mode = Absolute
[2019-11-16 22:57:04.694] >> ClearButtonMap
[2019-11-16 22:57:04.694] Pen button map cleared!
[2019-11-16 22:57:04.694] >> ButtonMap 1 "MOUSE1"
[2019-11-16 22:57:04.694] Pen button 1 mapped to 'MOUSE1'
[2019-11-16 22:57:04.694] >> ButtonMap 2 "MOUSE2"
[2019-11-16 22:57:04.694] Pen button 2 mapped to 'MOUSE2'
[2019-11-16 22:57:04.694] >> ButtonMap 3 "MOUSE3"
[2019-11-16 22:57:04.694] Pen button 3 mapped to 'MOUSE3'
[2019-11-16 22:57:04.694] >> ClearAuxButtonMap
[2019-11-16 22:57:04.694] Aux button map cleared!
[2019-11-16 22:57:04.694] >> PressureSensitivity 0
[2019-11-16 22:57:04.694] Tablet pressure sensitivity = 0.00000
[2019-11-16 22:57:04.694] >> PressureDeadzone 0
[2019-11-16 22:57:04.694] Tablet pressure deadzone = 0.00
[2019-11-16 22:57:04.694] >> ScrollSensitivity 0.5
[2019-11-16 22:57:04.694] Tablet scroll sensitivity = 0.50 scrolls per millimeter
[2019-11-16 22:57:04.694] >> ScrollAcceleration 1
[2019-11-16 22:57:04.694] Tablet scroll acceleration = 1.00
[2019-11-16 22:57:04.694] >> ScrollStopCursor false
[2019-11-16 22:57:04.694] Tablet scroll stop cursor = false
[... 22 lines cut ...]
[2019-11-16 22:57:04.695]   Report Id = 0x02
[2019-11-16 22:57:04.695]   Report Length = 8 bytes
[2019-11-16 22:57:04.695]   Detect Mask = 0x80
[2019-11-16 22:57:04.695]   Ignore Mask = 0x00
[... cut ...]
[2019-11-16 22:57:04.695] >> Start
[2019-11-16 22:57:04.702] TabletDriver started!
[2019-11-16 22:57:04.702] [STATUS] TABLET XP-Pen G640
[2019-11-16 22:57:04.702] [STATUS] HID 28BD 0094 FF0A 0001
[2019-11-16 22:57:04.702] [STATUS] WIDTH 160.00000
[2019-11-16 22:57:04.702] [STATUS] HEIGHT 100.00000
[2019-11-16 22:57:04.702] [STATUS] MAX_X 32000
[2019-11-16 22:57:04.702] [STATUS] MAX_Y 20000
[2019-11-16 22:57:04.702] [STATUS] MAX_PRESSURE 8191
[2019-11-16 22:57:04.702] [STATUS] AUX_BUTTONS 0
[2019-11-16 22:57:04.702] >> Log Off

SBAPKat, 2019-11-16:
The tablet seems to be recognized correctly, have you tried changing input mode to relative/ Windows ink? You should also make sure you've either uninstalled any other driver, or made sure it wasn't running in the background.

tillann, 2019-11-16:
Changing the mode doesn't seem to help. Don't have any drivers running in the background either.

SBAPKat, 2019-11-17:
Just to make sure, once you've changed the mode, have you restarted the driver with the appropriate button?

tillann, 2019-11-17:
That seems to make it work, atleast for Relative. But it seems quite choppy

SBAPKat, 2019-11-17:
So, if you take an area, relative mode will take up the corresponding sensivity, so if you have a very big area, it might feel quite weird. You could also try Windows ink mode, wich works in absolute tracking, but is not recommended for osu! due to its input lag while hovering.

SBAPKat, 2019-11-17:
For absolute(osu!) tracking, make sure the screen map is covered to its maximum, with the area selec to your monitor. You can then force the aspect ratio in the tablet area and restart the driver with the button.

tillann, 2019-11-17:
This worked, thanks a lot!

SBAPKat, 2019-11-17:
Glad to hear that. Could you please close this issue? 
Have a nice day!

closed 2019-11-17 by tillann