hawku/TabletDriver #535: Add please support XP-PEN 03 PRO

Seitry, 2019-11-18:
I want to use this driver, but for some reason the cursor does not respond to touches even when I deleted the old driver.
Add support XP-PEN 03 PRO

Thanks in advance for your reply.

SBAPKat, 2019-11-21:
Hi, it is normal for your cursor to boot love, as this tablet is not officially supported. 

Have you installed the vmulti driver? If so, please tick the debug checkbox in the console tab, and restart the driver, please send me the debug log once that's done, so I can help adding so support for this tablet.

Seitry, 2019-11-23:
> Hi, it is normal for your cursor to not move, as this tablet is not officially supported.
> 
> Have you installed the vmulti driver? If so, please tick the debug checkbox in the console tab, and restart the driver, please send me the debug log once that's done, so I can help adding support for this tablet.

Seitry, 2019-11-23:
[2019-11-23 18:08:46.099] [DEBUG] [HIDDevice] HID Device: Vendor: 'Logitech' Product: 'G102 Prodigy Gaming Mouse', Serial: '0571384B3336'
[2019-11-23 18:08:46.099] [DEBUG] [HIDDevice]   Vendor Id: 0x046D, Product Id: 0xC084
[2019-11-23 18:08:46.099] [DEBUG] [HIDDevice]   Usage Page: 0xFF00, Usage: 0x0002
[2019-11-23 18:08:46.099] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 20, OutputLen: 20
[2019-11-23 18:08:46.099] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.100] [DEBUG] [HIDDevice] HID Device: Vendor: 'UC-Logic' Product: 'star03 Pro', Serial: ''
[2019-11-23 18:08:46.100] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0077
[2019-11-23 18:08:46.100] [DEBUG] [HIDDevice]   Usage Page: 0xFF0A, Usage: 0x0001
[2019-11-23 18:08:46.100] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 8, OutputLen: 8
[2019-11-23 18:08:46.100] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.102] [DEBUG] [HIDDevice] HID Device: Vendor: 'SINO WEALTH' Product: 'Mechanical Keyboard', Serial: ''
[2019-11-23 18:08:46.102] [DEBUG] [HIDDevice]   Vendor Id: 0x0603, Product Id: 0x00F5
[2019-11-23 18:08:46.102] [DEBUG] [HIDDevice]   Usage Page: 0x0001, Usage: 0x0080
[2019-11-23 18:08:46.102] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 2, OutputLen: 0
[2019-11-23 18:08:46.102] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.103] [DEBUG] [HIDDevice] HID Device: Vendor: 'UC-Logic' Product: 'star03 Pro', Serial: ''
[2019-11-23 18:08:46.103] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0077
[2019-11-23 18:08:46.103] [DEBUG] [HIDDevice]   Usage Page: 0xFF01, Usage: 0x0000
[2019-11-23 18:08:46.103] [DEBUG] [HIDDevice]   FeatureLen: 6, InputLen: 0, OutputLen: 0
[2019-11-23 18:08:46.103] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.104] [DEBUG] [HIDDevice] HID Device: Vendor: 'SINO WEALTH' Product: 'Mechanical Keyboard', Serial: ''
[2019-11-23 18:08:46.104] [DEBUG] [HIDDevice]   Vendor Id: 0x0603, Product Id: 0x00F5
[2019-11-23 18:08:46.104] [DEBUG] [HIDDevice]   Usage Page: 0x000C, Usage: 0x0001
[2019-11-23 18:08:46.104] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 3, OutputLen: 0
[2019-11-23 18:08:46.104] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.106] [DEBUG] [HIDDevice] HID Device: Vendor: 'SINO WEALTH' Product: 'Mechanical Keyboard', Serial: ''
[2019-11-23 18:08:46.106] [DEBUG] [HIDDevice]   Vendor Id: 0x0603, Product Id: 0x00F5
[2019-11-23 18:08:46.106] [DEBUG] [HIDDevice]   Usage Page: 0xFF00, Usage: 0x0001
[2019-11-23 18:08:46.106] [DEBUG] [HIDDevice]   FeatureLen: 6, InputLen: 0, OutputLen: 0
[2019-11-23 18:08:46.106] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.107] [DEBUG] [HIDDevice] HID Device: Vendor: 'UC-Logic' Product: 'star03 Pro', Serial: ''
[2019-11-23 18:08:46.107] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0077
[2019-11-23 18:08:46.107] [DEBUG] [HIDDevice]   Usage Page: 0x000C, Usage: 0x0001
[2019-11-23 18:08:46.107] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 3, OutputLen: 0
[2019-11-23 18:08:46.107] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.108] [DEBUG] [HIDDevice] HID Device: Vendor: 'UC-Logic' Product: 'star03 Pro', Serial: ''
[2019-11-23 18:08:46.108] [DEBUG] [HIDDevice]   Vendor Id: 0x28BD, Product Id: 0x0077
[2019-11-23 18:08:46.108] [DEBUG] [HIDDevice]   Usage Page: 0x0001, Usage: 0x0080
[2019-11-23 18:08:46.108] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 2, OutputLen: 0
[2019-11-23 18:08:46.108] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.109] [DEBUG] [HIDDevice] HID Device: Vendor: 'Logitech' Product: 'G102 Prodigy Gaming Mouse', Serial: '0571384B3336'
[2019-11-23 18:08:46.109] [DEBUG] [HIDDevice]   Vendor Id: 0x046D, Product Id: 0xC084
[2019-11-23 18:08:46.109] [DEBUG] [HIDDevice]   Usage Page: 0x000C, Usage: 0x0001
[2019-11-23 18:08:46.109] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 5, OutputLen: 0
[2019-11-23 18:08:46.109] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.110] [DEBUG] [HIDDevice] HID Device: Vendor: 'Logitech' Product: 'G102 Prodigy Gaming Mouse', Serial: '0571384B3336'
[2019-11-23 18:08:46.110] [DEBUG] [HIDDevice]   Vendor Id: 0x046D, Product Id: 0xC084
[2019-11-23 18:08:46.110] [DEBUG] [HIDDevice]   Usage Page: 0x0001, Usage: 0x0080
[2019-11-23 18:08:46.110] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 2, OutputLen: 0
[2019-11-23 18:08:46.110] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice] HID Device: Vendor: 'Logitech' Product: 'G102 Prodigy Gaming Mouse', Serial: '0571384B3336'
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Vendor Id: 0x046D, Product Id: 0xC084
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Usage Page: 0xFF00, Usage: 0x0001
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 7, OutputLen: 7
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device', Serial: '123123123'
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Vendor Id: 0x00FF, Product Id: 0xBACC
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Usage Page: 0x0000, Usage: 0x000E
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   FeatureLen: 3, InputLen: 0, OutputLen: 0
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device', Serial: '123123123'
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Vendor Id: 0x00FF, Product Id: 0xBACC
[2019-11-23 18:08:46.111] [DEBUG] [HIDDevice]   Usage Page: 0x000D, Usage: 0x0001
[... cut ...]
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 65, OutputLen: 65
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device', Serial: '123123123'
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]   Vendor Id: 0x00FF, Product Id: 0xBACC
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]   Usage Page: 0xFF00, Usage: 0x0002
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]   FeatureLen: 0, InputLen: 33, OutputLen: 65
[2019-11-23 18:08:46.112] [DEBUG] [HIDDevice]

Seitry, 2019-11-27:
Hey, how's it going to add a tablet to the list?