hawku/TabletDriver #537: system error driver wont run

aaaaaaaaahhhhhhh, 2019-11-24:
if I try to run I get a system error: MSVCP140.dll not found and VCRUNTIME140.dll was not found. I've uninstalled and reinstalled the driver, I did a system check and everything was fine, I'm
 not sure what's wrong, I haven't downloaded anything or changed anything.

## Hardware/software information
Windows Version: 
windows 10 version 1909
Tablet Driver version: 
V0.2.3
Tablet model: 
wacom ctl-490

[image]
[image]

InfinityGhost, 2019-11-28:
Follow installation instructions thoroughly.

aaaaaaaaahhhhhhh, 2019-11-30:
> Follow installation instructions thoroughly.

I have multiple times. I didn't change anything when the driver stopped working, so it was installed properly before the error occurred

euphoricals, 2019-12-03:
I'm getting the same issue, and I've followed all of the instructions multiple times.

InfinityGhost, 2019-12-03:
Install both
- https://www.microsoft.com/net/download/dotnet-framework-runtime
- https://aka.ms/vs/15/release/vc_redist.x86.exe

euphoricals, 2019-12-03:
That worked for me, thanks so much

aaaaaaaaahhhhhhh, 2019-12-03:
> Install both
> 
> * https://www.microsoft.com/net/download/dotnet-framework-runtime
> * https://aka.ms/vs/15/release/vc_redist.x86.exe

it worked, thank you. I already repaired my C++ but your link work for some reason

closed 2019-12-03 by aaaaaaaaahhhhhhh