hawku/TabletDriver #552: Cursor stuck on left side of screen, tried searching issue here none of the solutions worked.

Ellenoria, 2019-12-19:
This started happening about a week ago.
While playing osu, tablet stopped working correctly, restarted driver and it failed saying that the vmulti was missing. Uninstalled and reinstalled Vmulti_64.
Since then the cursor has been stuck on the left side whenever I try to select osu as the window. Selecting anything else and changing the active region of the tablet is very unstable and the cursor lags behind.
I've deleted the whole zip, uninstalled the drivers, then reinstalled everything and that didn't work. 
I've tried disabling raw input for osu and that works for about 10 seconds before snapping to the left side of the screen again. 
There's no option on the hawku driver to disable windows ink and it won't let me disable it from regedit either, I tried disabling it from there as well. 

I'd like to know if there's something I'm not seeing that I'm doing incorrectly. The vmulti driver should be the only driver installed. 
Any other driver makes osu unplayable because it lags just enough for me to not be able to play, so I'd really appreciate if someone could look over this. 

These are the settings that I was working with earlier and they were working fine up until the issue.

[image]

Some console logs

[image]

[image]

[image]

## Hardware/software information
Windows Version: windows 10 home latest update

Tablet Driver version: 0.2.3

Tablet model: veikk s640

Ellenoria, 2019-12-23:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3995767/startuplog.txt)

Since posting I've also tried installing and uninstalling wacom drivers, installing a previous version of this driver, and changing the configs detect mask.

I have a temporary fix by disabling my second monitor and restarting my computer, but I'd really not like to restart my computer and have a single monitor whenever I feel like playing the game.

Ellenoria, 2019-12-25:
I've found the main issue.
When setting it to the osu window the driver isn't recognizing that it's on the screen anymore since it's minimized, on previous versions of windows it still pulled the window (still showed the same offset on the driver gui which is why I wasn't sure what was going on)
In game you have to temporarily set the cursor to never be confined to the screen, then change the tablet to the osu window and it will work like it did before.
This issue was driving me insane for weeks so hopefully this solution helps someone else.

closed 2019-12-25 by Ellenoria

youi645, 2021-06-25:
Can you explain more on how to fix it Im confused.

youi645, 2021-06-26:
ity didnt hewlp :)))()((((

Ellenoria, 2021-06-26:
This is almost a two year old post. Most likely your solution isnt here and you’ll have to make a new one.