hawku/TabletDriver #559: g430s_B not changing area

LegoshiKun, 2019-12-26:
I have an XP-Pen G430S_B. I already did the thing in the config to make it detectable. It detects it fine, but it won't change the area of the tablet no matter what I do. No errors in the driver console. I did uninstall and reinstall, and restarted my computer to no avail. I don't have the XP-Pen drivers installed. I did notice a Pentablet HID device in devmgmt but upon disabling it I discovered it's the driver for hawku.
So I'm basically stumped here, will switch to XP-Pen drivers until I can find a fix.

## Hardware/software information
Windows Version: Windows 10 Home, NT 6.2.9200.0

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G430S_B

LegoshiKun, 2019-12-26:
Here's the changes I made to tablet.cgf:
On lines 653-657:
`#
# XP-Pen G430S
#
HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
CheckString Product "4 inch PenTablet"
Name "XP-Pen G430S_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 10160
MaxY 7620
MaxPressure 8191
Width 101.6
Height 76.2
#InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00`

LegoshiKun, 2019-12-26:
Here are the changes I made to tablet.cfg:

_HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
CheckString Product "4 inch PenTablet"
Name "XP-Pen G430S_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 10160
MaxY 7620
MaxPressure 8191
Width 101.6
Height 76.2
#InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00_

LegoshiKun, 2019-12-26:
Ok apparently uninstalling vmulti worked?????????

closed 2019-12-26 by LegoshiKun

uyopi, 2020-01-12:
Would you mind elaborating on what exactly you did to resolve this issue? Same thing is happening to me, but unistalling/reinstalling VMulti doesn't seem to help.

LegoshiKun, 2020-01-12:
I uninstalled vmulti and never reinstalled it. I also have pentablet, xp-pen's drivers, installed (though I'm not sure if that helps or not).

uyopi, 2020-01-12:
Nope, installing XP-Pen's drivers doesn't seem to help either. While having them installed makes hawku's detect my tablet even without the additional VMulti, it still doesn't let me change active area/make filters workable. Having both of them run at the same time only uses Pentablet's configuration. It's as if hawku's drivers just aren't able to overwrite default Windows' way of handling tablets.