hawku/TabletDriver #569: Can't change active area

uyopi, 2020-01-10:
I own a G430S, the B version. Since the drivers couldn't detect my tablet and kept thinking I have a Wacom, I used the fix from #520 . Now, while the drivers do recognize my tablet, it seems like I can't change the active area. It stays full no matter what I select.

Here are the logs:
[2020-01-10 08:44:22.112] \ Reading 'config/init.cfg'
[2020-01-10 08:44:22.112] >> Log startuplog.txt
[2020-01-10 08:44:22.112] Log file 'startuplog.txt' opened.
[2020-01-10 08:44:22.112] >> LogDirect true
[2020-01-10 08:44:22.112] Log direct print = True
[2020-01-10 08:44:22.112] >> Include "config\tablet.cfg"
[2020-01-10 08:44:22.112] \ Reading 'config\tablet.cfg'
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> # Example tablet definition:
[2020-01-10 08:44:22.112] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> # VID: 0x056a
[2020-01-10 08:44:22.112] >> # PID: 0x00dd
[2020-01-10 08:44:22.112] >> # HID Usage Page: 0x0D
[2020-01-10 08:44:22.112] >> # HID Usage: 0x01
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> # Wacom CTL-470 (Wacom drivers installed)
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2020-01-10 08:44:22.112] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2020-01-10 08:44:22.112] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2020-01-10 08:44:22.112] >> ReportId 0x02
[2020-01-10 08:44:22.112] >> ReportLength 11
[2020-01-10 08:44:22.112] >> DetectMask 0x40
[2020-01-10 08:44:22.112] >> MaxX 14720
[2020-01-10 08:44:22.112] >> MaxY 9200
[2020-01-10 08:44:22.112] >> MaxPressure 1023
[2020-01-10 08:44:22.112] >> Width 147.2
[2020-01-10 08:44:22.112] >> Height 92.0
[2020-01-10 08:44:22.112] >> TabletFormat WacomDrivers
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> # Wacom CTL-470 
[2020-01-10 08:44:22.112] >> #
[2020-01-10 08:44:22.112] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2020-01-10 08:44:22.128] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-01-10 08:44:22.128] >> Name "Wacom CTL-470"
[2020-01-10 08:44:22.128] >> ReportId 0x02
[2020-01-10 08:44:22.128] >> ReportLength 10
[2020-01-10 08:44:22.128] >> DetectMask 0x40
[2020-01-10 08:44:22.128] >> MaxX 14720
[2020-01-10 08:44:22.128] >> MaxY 9200
[2020-01-10 08:44:22.128] >> MaxPressure 1023
[2020-01-10 08:44:22.128] >> Width 147.2
[2020-01-10 08:44:22.128] >> Height 92.0
[2020-01-10 08:44:22.128] >> InitFeature 0x02 0x02
[2020-01-10 08:44:22.128] >> #
[2020-01-10 08:44:22.128] >> # Wacom CTH-470 (Wacom drivers installed)
[2020-01-10 08:44:22.128] >> #
[2020-01-10 08:44:22.128] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2020-01-10 08:44:22.128] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2020-01-10 08:44:22.128] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2020-01-10 08:44:22.128] >> ReportId 0x02
[2020-01-10 08:44:22.128] >> ReportLength 11
[2020-01-10 08:44:22.128] >> DetectMask 0x40
[2020-01-10 08:44:22.128] >> MaxX 14720
[2020-01-10 08:44:22.128] >> MaxY 9200
[2020-01-10 08:44:22.128] >> MaxPressure 1023
[2020-01-10 08:44:22.128] >> Width 147.2
[2020-01-10 08:44:22.128] >> Height 92.0
[2020-01-10 08:44:22.128] >> TabletFormat WacomDrivers
[2020-01-10 08:44:22.128] >> #
[2020-01-10 08:44:22.128] >> # Wacom CTH-470
[2020-01-10 08:44:22.128] >> #
[2020-01-10 08:44:22.128] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2020-01-10 08:44:22.128] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-01-10 08:44:22.128] >> Name "Wacom CTH-470"
[... 686 lines cut ...]
[2020-01-10 08:44:22.190] 
[2020-01-10 08:44:22.190] Area:
[2020-01-10 08:44:22.190]   Desktop = 3840px x 2160px
[2020-01-10 08:44:22.190]   Screen Map = 3840px x 2160px @ X+0px, Y+0px
[2020-01-10 08:44:22.190]   Tablet =  85.00mm x 47.81mm @ Min(50.80mm, 28.58mm) Max(135.80mm, 76.39mm)
[2020-01-10 08:44:22.190]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2020-01-10 08:44:22.190] 
[2020-01-10 08:44:22.190] >> Start
[2020-01-10 08:44:22.190] TabletDriver started!
[2020-01-10 08:44:22.190] [STATUS] TABLET XP-Pen G430S_B
[2020-01-10 08:44:22.190] [STATUS] HID 28BD 0913 FF0A 0001
[2020-01-10 08:44:22.190] [STATUS] WIDTH 101.60000
[2020-01-10 08:44:22.190] [STATUS] HEIGHT 76.20000
[2020-01-10 08:44:22.190] [STATUS] MAX_X 10160
[2020-01-10 08:44:22.190] [STATUS] MAX_Y 7620
[2020-01-10 08:44:22.190] [STATUS] MAX_PRESSURE 8191
[2020-01-10 08:44:22.190] [STATUS] AUX_BUTTONS 0
[2020-01-10 08:44:22.190] >> Log Off

## Hardware/software information
Windows Version: Windows 10 Home Edition

Tablet Driver version: 0.2.3

Tablet model: XP-Pen Star G430S_B

uyopi, 2020-01-12:
After a long, long time trying different things, it seems like the problem lies within the hawku drivers not starting a service controlling the tablet. Is there any way to force that?

uyopi, 2020-01-13:
Using driver version 0.2.2 instead of 0.2.3 fixed the issue for me. Closing.

closed 2020-01-13 by uyopi

TempleRxse, 2020-01-31:
did not work for me. Any help plz!?

HyNek0, 2020-09-13:
Idk if you found this by yourself since it's been a while, but changing the InitReport values as stated in [this post](https://www.reddit.com/r/osugame/comments/du7689/xppen_g430s_b_tabletdriver_fix/) works wonders.

TempleRxse, 2020-09-14:
yes i found that, but thanks anyways.