hawku/TabletDriver #576: "save settings" button has issue

aDelightz, 2020-01-24:
[image]
idk how to give it perms for editing file directory
this means id have to export and import settings manually through these options 
[image]
 every time and select the files in /documents but by then it would be quicker to just do the settings again

## Hardware/software information
Windows Version: Windows 10, version 1909

Tablet Driver version: TabletDriverV0.2.3

Tablet model: VEIKK S640

aDelightz, 2020-02-09:
fixed by reinstalling: startuplog text file created automatically this time

aDelightz, 2020-02-09:
fixed by putting it in a separate folder

Jeromvjr, 2020-03-04:
wait how did you exactly fix it, I have the same problem

aDelightz, 2020-03-27:
i originally put it in E:\Program Files folder which the app doesn't have the permissions to change files in, so I moved it from the program files folder straight to the ssd E:\ as its own folder (if you are putting it on your main harddrive/ssd for you it will probably be C:\)
moving the tablet drivers folder to documents or desktop or whatever would probably work as well so long as when you try to click save settings it doesn't pop up with a message saying it cannot create and edit files there.

sorry for the late response xd

billskii, 2020-05-24:
I have the same problem

slikaa, 2020-12-15:
same i need help

aDelightz, 2022-04-30:
oops i never closed this

closed 2022-04-30 by aDelightz