hawku/TabletDriver #578: CTL-480 Cursor issue

StaticRaider, 2020-01-28:
I bought a Wacom CTL-480 last November and it was fine until a week ago but it only does it on my main computer, I tried it on my sisters with the same drivers and it was fine its only my main computer that it doesn't like. Its not a osu issue because it happens on the desktop and i have run out of ideas on how to fix it, One thing i did realise was at the top it says (Wacom drivers installed) when i haven't installed any wacom drivers i only use these drivers. The issue is when i am trying to use the tablet the cursor will teleport to a different location then teleport back, I also own a XP-pen G640 which works fine so i thought it was a bit strange that the Wacom tablet doesn't like my computer. Any help will be much appreciated. 

## Hardware/software information
Windows Version: 18362.592

Tablet Driver version: 0.2.3

Tablet model: CTL-480

hqwizard, 2020-02-05:
i think i have a similar issue with my ctl-480 as well, although mine occasionally works. can you record some footage?

StaticRaider, 2020-02-06:
Sorry for the late reply, I have uploaded a video of the issue i am having. https://www.youtube.com/watch?v=S3yeXDlfLZ0&feature=youtu.be

hqwizard, 2020-02-11:
no problem, my issue was cursor rapidly going everywhere no matter how i move the pen, in some places on the tablet it could jump to three different points on screen. I didn't encounter this problem since i removed my 0.2.3 driver left from reinstalling windows, then downloading a fresh 0.2.3 one (so yeah they're actually the same but i did it anyway). also i never turned on "run at windows startup" after that. maybe that could help but i don't know

if this issue appears again i'm going to record it for you

closed 2020-02-14 by StaticRaider

hqwizard, 2020-02-17:
in the tools folder inside driver's folder there is a DisableWacomDrivers.bat
i uninstalled hawku drivers, ran it, then installed again and it fixed my issue