hawku/TabletDriver #580: cant change active are on g430s_B

TempleRxse, 2020-01-31:
so I did use #520 to fix hawku recognising the tablet, and I also use #569 and changed from 0.2.3 to 0.2.2, but I still cant change active area. Any else I need to do?

## Hardware/software information
Windows Version: 10 32bit

Tablet Driver version: 0.2.2

Tablet model: g430s_B

uyopi, 2020-02-01:
Use [https://www.reddit.com/r/osugame/comments/du7689/xppen_g430s_b_tabletdriver_fix/](url) instead of the one from #520. The only difference is a few more digits in the Init Report, but it worked wonders for me.

HyNek0, 2020-04-09:
This worked for me too ! I had to uninstall and reinstall everything, but using the values in InitReport from [this post](https://www.reddit.com/r/osugame/comments/du7689/xppen_g430s_b_tabletdriver_fix/) made it work without anything else running !