hawku/TabletDriver #583: драйвер не может найти планшет xp-pen g 430s

Gabb1Max, 2020-02-03:
перепробовал все уже , не могу использовать драйвер  , пишет что планшет не найден в консоли 

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

NiceGoodRU, 2020-02-03:
вставь в конфиг

#
# XP-Pen G430S_B
#
HIDTablet 0x28BD 0x0913 0xFF0A 0x0001
CheckString Product "4 inch PenTablet"
Name "XP-Pen G430S_B"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 10160
MaxY 7620
MaxPressure 8191
Width 101.6
Height 76.2
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

Gabb1Max, 2020-02-03:
а как , если можешь объясни плз , а то у меня файл конфиг не открывается

NorkKiss, 2020-12-25:
пишет что не найден планшет 
star 03 v2 
помоги пожалуйста.

picleric, 2021-07-09:
у меня планшет ctl 472 и у меня после скачивания hawku все работало нормально но на следующий день драйвер начал тупить и  у меня не меняется область а я лешва и режим левши у меня тоже не работает 
hawku я переустонавливал несколько раз ничего не помогает

BlackHoliday, 2022-03-29:
помогите пожалуйста драйвер hawku не читает мой планшет parblo a640 v2, что нужно вставить в конфиг?