hawku/TabletDriver #599: Buttons not working in Windows Ink Mode

thomasaull, 2020-02-26:
First, I have to say these drivers are awesome. The wacom ones were driving me nuts, so I looked up an alternative and this one looks really promising so far (I'm not at all playing osu, but rather using my pen tablet as a more ergonomic mouse alternative. I noticed that, when using Windows Ink mode, I couldn't use the middle/right mouse button settings anymore (Scroling and custom Keymaps did work!). Is this a known limitation? Do I have to change any settings in Windows or might this be a bug?

## Hardware/software information
Windows Version: 10.0.19041

Tablet Driver version: 0.2.3

Tablet model: Wacom CTH-490

VinhDevNguyen, 2020-04-23:
Same here :<
I can't use this driver for Blender because without middle mouse button, I can't zoom or rotate.

thomasaull, 2020-04-23:
@VinhDevNguyen Try setting Area > Mode to Absolute. This way the settings in Buttons are working!

VinhDevNguyen, 2020-04-23:
It's work but I can't use pen pressure feature

thomasaull, 2020-04-23:
You're right… Looks like you're stuck with the official wacom driver in this case … :(