hawku/TabletDriver #602: XP-Pen Star G640 not being recognized

laurorual, 2020-03-14:
Hi everyone!

I'm trying to install this driver since reddit told me it has less input lag but i cant make the TabletDriverGUI recognize my tablet, it always says in the console "tablet not found!".

First i tried uninstalling all XP-Pen drivers, rebooting PC, installing hawku drivers through "install_vmulti_driver.bat" and open TabletDriverGUI. Not working.
Then, after running "remove_vmulti_driver.bat", i tried the "install_vmulti_64.bat" inside the "driver_vmulti" folder, still not working.

Then i tried with XP-Pen drivers, both "normal" and "osu" drivers, still not working, even if i end "Pentable Service" process in the Task Manager. Everything i tried i get "tablet not found!"

It is not a issue with my tablet since it works fine with official XP-Pen drivers, so i don't know what it could be.
Maybe i have a new revision of the G640?

## Hardware/software information
Windows Version: 1909

Tablet Driver version: 0.2.3

Tablet model: XP-Pen Star G640

Freemmm, 2020-03-19:
I have the same problem...

NotOnAClient, 2020-04-01:
Same problem here, might be similar to the g430s_B where the new versions wouldn't work.

HongClarence, 2020-04-01:
Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad anything similar and paste this
#
# XP-Pen G640
#
HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 64000
MaxY 40000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

NotOnAClient, 2020-04-01:
> Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad or anything similar and paste this
> 
> # XP-Pen G640
> HIDTablet 0x28BD 0x0914 0xFF0A 0x0001
> #CheckString Manufacturer "XP-PEN"
> CheckString Product "6 inch PenTablet"
> Name "XP-Pen G640"
> ReportId 0x02
> ReportLength 12
> DetectMask 0x80
> MaxX 16000
> MaxY 10000
> MaxPressure 8191
> Width 160.0
> Height 100.0
> InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
> 
> or just replace the tablet.cfg file with this, remember to change the txt file into cfg.
> [tablet.txt](https://github.com/hawku/TabletDriver/files/4414588/tablet.txt)

Thanks! It works now

NotOnAClient, 2020-04-01:
> Thanks! It works now

I take that back, it detects the tablet but the cursor does not move.

HongClarence, 2020-04-01:
Try reinstalling the vmulti drivers, it should work, make sure you uninstall xp-pen's drivers as well

NotOnAClient, 2020-04-01:
Thanks it works now

thunderboltcg, 2020-04-11:
omggggggggggggggg thanks soooooooooooooo much

HongClarence, 2020-04-11:
No problem, do let me know if there is any latency though, not sure if its my tablet or just me

Hexishu, 2020-04-20:
@HongClarence It doesn't work for me (or my friend) though, I tried to apply your fix but it still says "Driver stopped. Restarting! Check console.

2020-04-21 00:57:42 [INFO] >> Name "Wacom CTH-461"
2020-04-21 00:57:42 [INFO] >> ReportLength 9
2020-04-21 00:57:42 [INFO] >> DetectMask 0x40
2020-04-21 00:57:42 [INFO] >> MaxX 14720
2020-04-21 00:57:42 [INFO] >> MaxY 9200
2020-04-21 00:57:42 [INFO] >> MaxPressure 1023
2020-04-21 00:57:42 [INFO] >> Width 147.200
2020-04-21 00:57:42 [INFO] >> Height 92.000
2020-04-21 00:57:42 [INFO] >> InitFeature 0x02 0x02
2020-04-21 00:57:42 [INFO] >> # Wacom CTH-470
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x00DE 0xFF0D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x00DE 0xFF0D 0x0001
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x00DE 0x000D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x00DE 0x000D 0x0001
2020-04-21 00:57:42 [INFO] >> Name "Wacom CTH-470"
2020-04-21 00:57:42 [INFO] >> ReportLength 10
2020-04-21 00:57:42 [INFO] >> DetectMask 0x40
2020-04-21 00:57:42 [INFO] >> MaxX 14720
2020-04-21 00:57:42 [INFO] >> MaxY 9200
2020-04-21 00:57:42 [INFO] >> MaxPressure 1023
2020-04-21 00:57:42 [INFO] >> Width 147.200
2020-04-21 00:57:42 [INFO] >> Height 92.000
2020-04-21 00:57:42 [INFO] >> InitFeature 0x02 0x02
2020-04-21 00:57:42 [INFO] >> # Wacom CTH-480
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x0302 0xFF0D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x0302 0xFF0D 0x0001
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x0302 0x000D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x0302 0x000D 0x0001
2020-04-21 00:57:42 [INFO] >> Name "Wacom CTH-480"
2020-04-21 00:57:42 [INFO] >> ReportLength 10
2020-04-21 00:57:42 [INFO] >> DetectMask 0x40
2020-04-21 00:57:42 [INFO] >> MaxX 15200
2020-04-21 00:57:42 [INFO] >> MaxY 9500
2020-04-21 00:57:42 [INFO] >> MaxPressure 1023
2020-04-21 00:57:42 [INFO] >> Width 152.000
2020-04-21 00:57:42 [INFO] >> Height 95.000
2020-04-21 00:57:42 [INFO] >> InitFeature 0x02 0x02
2020-04-21 00:57:42 [INFO] >> # Wacom CTH-490
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x033C 0xFF0D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x033C 0xFF0D 0x0001
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x033C 0x000D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x033C 0x000D 0x0001
2020-04-21 00:57:42 [INFO] >> Name "Wacom CTH-490"
2020-04-21 00:57:42 [INFO] >> ReportLength 10
2020-04-21 00:57:42 [INFO] >> DetectMask 0x40
2020-04-21 00:57:42 [INFO] >> MaxX 15200
2020-04-21 00:57:42 [INFO] >> MaxY 9500
2020-04-21 00:57:42 [INFO] >> MaxPressure 2047
2020-04-21 00:57:42 [INFO] >> Width 152.000
2020-04-21 00:57:42 [INFO] >> Height 95.000
2020-04-21 00:57:42 [INFO] >> InitFeature 0x02 0x02
2020-04-21 00:57:42 [INFO] >> Type WacomIntuos
2020-04-21 00:57:42 [INFO] >> # Wacom CTH-661
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x00D3 0xFF0D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x00D3 0xFF0D 0x0001
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x00D3 0x000D 0x0001
2020-04-21 00:57:42 [WARNING] Can't open HID tablet 0x056A 0x00D3 0x000D 0x0001
2020-04-21 00:57:42 [INFO] >> Name "Wacom CTH-661"
2020-04-21 00:57:42 [INFO] >> ReportLength 9
2020-04-21 00:57:42 [INFO] >> DetectMask 0x40
2020-04-21 00:57:42 [INFO] >> MaxX 21648
2020-04-21 00:57:42 [INFO] >> MaxY 13700
2020-04-21 00:57:42 [INFO] >> MaxPressure 1023
2020-04-21 00:57:42 [INFO] >> Width 216.480
2020-04-21 00:57:42 [INFO] >> Height 137.000
2020-04-21 00:57:42 [INFO] >> InitFeature 0x02 0x02
2020-04-21 00:57:42 [INFO] >> # Wacom CTH-661(A)
2020-04-21 00:57:42 [INFO] >> Tablet 0x056A 0x00D8 0xFF0D 0x0001
[... 207 lines cut ...]
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Usage Page: 0x000D, Usage: 0x0001
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 8, OutputLen: 0
2020-04-21 00:57:42 [HIDDevice] [DEBUG] 
2020-04-21 00:57:42 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0001
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 65, OutputLen: 65
2020-04-21 00:57:42 [HIDDevice] [DEBUG] 
2020-04-21 00:57:42 [HIDDevice] [DEBUG] HID Device: Vendor: 'DJP Inc.' Product: 'Virtual Multitouch Device'
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Vendor Id: 0x00FF, Product Id: 0xBACC
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   Usage Page: 0xFF00, Usage: 0x0002
2020-04-21 00:57:42 [HIDDevice] [DEBUG]   FeatureLen: 0, InputLen: 33, OutputLen: 65
2020-04-21 00:57:42 [HIDDevice] [DEBUG] 
2020-04-21 00:57:42 [INFO] 
2020-04-21 00:57:42 [INFO]   Driver version: 0.1.5.5 Devocub Edition
2020-04-21 00:57:42 [INFO]   Windows version: Microsoft Windows NT 6.2.9200.0
2020-04-21 00:57:42 [INFO]   Windows product: Windows 10 Home
2020-04-21 00:57:42 [INFO]   Windows release: 1909
2020-04-21 00:57:42 [INFO] 
2020-04-21 00:57:42 [INFO] >> CheckTablet
2020-04-21 00:57:42 [ERROR] Tablet not found!
2020-04-21 00:57:42 [ERROR] Check the list of supported tablets from the GitHub page.
2020-04-21 00:57:42 [ERROR] http://github.com/hawku/TabletDriver
Driver stopped. Restarting!

HongClarence, 2020-04-21:
Did you save the tablet.cfg file before trying? If you did and it still isn't working, try replacing the file with the on i posted

Hexishu, 2020-04-21:
Yes, I did everything and yes, I saved your file ofc. I am not dumb but
it's still not working unfortunately:/

HongClarence, 2020-04-21:
Hm, I am unable to see what's the problem. It just seems that your tablet isn't being detected even though all the id matches. Have you tried just uninstalling everything and redownloading all the files again?

Hexishu, 2020-04-21:
Yes, yes I did. Even uninstall vmulti and install vmulti several times,
deleted the driver files and then extracted them again and replaced the cfg
file with your one.

HongClarence, 2020-04-21:
Does your tablet work with the default driver by xp-pen? Sorry for the dumb questions just want to make sure that i get as much info as possible to help u

Hexishu, 2020-04-21:
Yea sure I know you don't mean it bad.

Um yes it does work with the xp pen drivers and I uninstalled them all. The
tablet is getting recognized as 6 inch PenTablet.

HongClarence, 2020-04-21:
Did you install the hardware update for the tablet? The one from xp-pen

Hexishu, 2020-04-21:
you mean Firmware update? I can try that

HongClarence, 2020-04-21:
Yup firmware update

schmoqz, 2020-04-22:
HongClarence Your a fucking G dawg, thanks so much, i tried getting it to work for no joke 10 hours
was about to reset my windows dude

Hexishu, 2020-04-23:
Did. But still the same result, not being recognized.

HongClarence, 2020-04-23:
Hi @Hexishu , sadly i am running out of ideas of what the problem might be since it works for others. I do notice a difference between our console debug, specifically this part.
2020-04-21 00:57:42 [HIDDevice] [DEBUG] HID Device: Vendor: 'UGTABLET' Product: '6 inch PenTablet'
Yours seems to lack a serial number as compared to mine,
[2020-04-23 23:58:37.478] [DEBUG] [HIDDevice] HID Device: Vendor: 'UGTABLET' Product: '6 inch PenTablet', Serial: '000000'
Did you get the tablet from the official xp-pen seller?

Hexishu, 2020-04-23:
Yes, from amazon.(speaking for a friend tho he has the tablet)

HongClarence, 2020-04-25:
The only other solution that i can think of is to contact the xp-pen customer support and tell them about the missing serial number, thought i doubt they will actually help since this driver is not by them

JOJOx408, 2020-05-07:
> Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad or anything similar and paste this
>
> # XP-Pen G640
> HIDTablet 0x28BD 0x0914 [...]

Thank you so much it finally works!

Ch3zZz, 2020-06-02:
> Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad or anything similar and paste this
>
> # XP-Pen G640
> HIDTablet 0x28BD 0x0914 [...]

thank you man

JustaPeriod, 2020-06-04:
@hawku @InfinityGhost @AbdShullah You should add this revision to your latest driver builds, can confirm that a friend of mine bought a new tablet off amazon and got this model of it, requiring the config above.

InfinityGhost, 2020-06-04:
> @hawku @InfinityGhost @AbdShullah You should add this revision to your latest driver builds, can confirm that a friend of mine bought a new tablet off amazon and got this model of it, requiring the config above.

afaik @AbdShullah already has added the revision to his repo, I haven't done anything because I'm no longer developing my fork in favor of my own project.

Psyite, 2020-06-27:
Hello! Thank you for your config changes, it has allowed me to use my tablet (G640) for osu! but, I have one problem, I'm getting cursor lag whenever I use this config with the hawku drivers. (I've tried everything, reinstalling drivers, tweaking settings like FPS and resolution in osu! etc,) The lag only occurs when I move my cursor slowly (like a small jump or spinner) my cursor would randomly jump and freeze making it extremely hard for me to do small jump sections on maps or do spinners, however I can do big jumps accurately. I have used my tablet with the XP-Pen Official drivers and it worked fine. (aside from it being Official drivers, my cursor would sway, but i could still somewhat play and there was no cursor lag) The closest fix I've had to this problem is experimenting with the smoothing filter, it has somewhat stopped the cursor lag but it feels too slow and clunky and it's not a 100% fix either, also, this problem isn't specifically an osu! problem (lag still occurs even if I'm not tabbed into osu!)

I can provide screenshots and recordings if need be, I have searched around the internet and couldn't find an accurate fix to my problem, it seems that I'm the only person with this problem as reviews on the tablet have not mentioned this problem. It's most annoying as I have previously used the Huion 420 with the hawku drivers and I had no cursor lag and gameplay felt perfect (switched for a bigger tablet with a better pen) 

I'm not sure if area size or anything like that matters but I've attached what I'm using now to play osu! (Area size, Smoothing filter settings, etc) I'd really appreciate it if you could help as this is making me pretty annoyed.

[Area size](https://gyazo.com/f8f20bb873f712a8fd14fd4672de9b4f)
[Smoothing Filter](https://gyazo.com/6de1ba778e6ef284a8cdca6bac5a13b3)
[PC Specs](https://gyazo.com/f55bf3a06b12ef6cb50c56f19351a088)

Psyite, 2020-06-28:
Okay, so I've found what the problem is. I experimented with changing my area size and presto! The glitching stopped, so I've switched from 7cm to 10cm and after getting used to it, I got some good plays.

Apologies if I wasted any ones time.

AvevoN, 2020-08-18:
> Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad or anything similar and paste this
>
> # XP-Pen G640
> HIDTablet 0x28BD 0x0914 [...]

Thank you so much!!!

rasters10, 2020-08-20:
> Hi, after poking around for 3 hours I finally got it to work. Head to the config folder, edit the tablet.cfg with notepad or anything similar and paste this
>
> # XP-Pen G640
> HIDTablet 0x28BD 0x0914 [...]

Thank you so much, i tried everything and it doesn't work, but after this my tablet with connected!

Frosteyy, 2020-09-22:
> 
> […](#)

you may have fixed it by now but if you didn't heres the line that fixed mine          
Tablet 0x28BD 0x0914 0xFF0A 0x0001
#CheckString Manufacturer "XP-PEN"
CheckString Product "6 inch PenTablet"
Name "XP-Pen G640"
ReportId 0x02
ReportLength 12
DetectMask 0x80
MaxX 16000
MaxY 10000
MaxPressure 8191
Width 160.0
Height 100.0
InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

What's different is that I removed HID from HIDTablet and it fixed it

Hexishu, 2020-09-28:
> > […](#)
> 
> you may have fixed it by now but if you didn't heres the line that fixed mine
> 
> # XP-Pen G640s
> Tablet 0x28BD 0x0906 0xFF0A 0x0001
> Name "XP-Pen G640s"
> ReportId 0x02
> ReportLength 10
> DetectMask 0x80
> MaxX 32000
> MaxY 20000
> MaxPressure 8191
> Width 160.0
> Height 100.0
> InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00
> 
> What's different is that I removed HID from HIDTablet and it fixed it for some reason

I'll try that for my friend ty in advance

Merlingers, 2021-07-04:
Hello, i have a problem with my XP-Pen G640.
I can't use the Absolute (osu!) mode with the settings from above but it works fine with the Relative(Mouse) mode.
So my Question now is do i need the absolute mode or is it something what isned nessesary to play osu.
and for me how can i fix it that i can play with it.
just recently switched from XP-Pen drivers to Hawku Driver. i deinstalled the previous driver.

joelchan617, 2021-08-16:
Hi guys, I know im kinda late with this issue but I've tried the debugs you guys have suggested but it just isn't working for me. Anyone can provide any help?