hawku/TabletDriver #604: Driver stopped. Restarting! Check console!!!

satorin209, 2020-03-20:
i haven't ever installed any tablet driver on my pc, but i still get those messages. have tried restarting pc, using another cable, plugging the tablet to different usb ports, reinstalling vmulti several times, but none of this ever helped.

## Hardware/software information
Windows Version: 10 x64

Tablet Driver version: 0.2.3

Tablet model: Wacom CTL-471
[image]

Kqmisqmq, 2020-04-14:
Same problem but with CTL-480

MuNgUmAn, 2020-10-20:
Same problem with the CTL-472

Kqmisqmq, 2020-10-20:
@MuNgUmAn i fixed it by putting the folder on my desktop... idk if it can help you (the folder where hawku drivers are)

V1vic, 2021-03-11:
> @MuNgUmAn i fixed it by putting the folder on my desktop... idk if it can help you (the folder where hawku drivers are)

Yea,none of that worked and im pissed cuz at first it worked  now it doesnt it just does the same thing error and error nothing works

V1vic, 2021-03-11:
> Same problem with the CTL-472

same,
[image]