hawku/TabletDriver #614: i can't install vmulti  driver

iiSqu4Qy, 2020-04-05:
## Hardware/software information
Windows Version: 7 

Tablet Driver version: 0.2.3

Tablet model: CTH-490

When i try to install vmulti driver it doesn't install anything and when i try to install vmulti 64 it just
https://gyazo.com/6293aff1a44008dcbd308e145a178abf

and the link to the xp site so you can download it induvidually doesn't work

Vlothx, 2020-05-07:
Hello, looks like Hawku stopped the development of this driver. I would recommend trying [this fork](https://github.com/InfinityGhost/TabletDriver). You might need to disable your antivirus, or if don't want to do that, you can compile it by yourself.

iiSqu4Qy, 2020-05-07:
i fixed it ;p I got a new computer cause the computer died