hawku/TabletDriver #615: "Can't open VMulti Device!"

Ducknorriss, 2020-04-06:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: (irrelevant) XPpen g640s

When I open the driver and check the console it says that the bin file cannot be found:

[image]

Because of this, I go to open the bin file, and when I do, this pops up for a quick second before closing:
[image]

When I open the bin file again, after downloading vmulti driver, I get the same error. I would like to add that I just got the laptop that I am using a week ago and have not downloaded any other drivers, so that has no interference. Anyone have any suggestions?

inevail, 2020-04-13:
exact same w me, have you figured out the issue?

Ducknorriss, 2020-04-14:
No, I have not. I will probably just try using different drivers.

inevail, 2020-04-14:
I found a fix. There should be a folder automatically on your desktop with all of the contents of the hawku driver. You have to use the TabletHub specifically from that folder.

inevail, 2020-04-14:
if there isn’t, create a whole new folder on your desktop and extract all contents of the hawku driver into it.

Ducknorriss, 2020-04-14:
what exactly do you mean by "TabletHub". I could not find a file by that or a similar name.

inevail, 2020-04-14:
Do you have discord? if so add me inevail#4935, i’ll be able to help you out

closed 2020-04-14 by Ducknorriss