hawku/TabletDriver #617: "Tablet Not found" (xp-pen G640)

Terdzz, 2020-04-08:
Since this morning , when i'm launching my tablet driver , it says that my tablet isnt found , like as if it was not supported by the driver but the xp-pen G640 Is supported so i dont know , and as i said i had no problem with the driver since 3 monce so i dont understand , can someone help ? 
[image]

thunderboltcg, 2020-04-11:
same problem someone help fast

Terdzz, 2020-04-11:
Okay finally i just , reinstalled it and it worked so :O

schmoqz, 2020-04-22:
What did you uninstall? I have same problem, please tell me everything you have uninstalled
and what you have installled

Vergonov, 2020-04-27:
I have the same issue.