hawku/TabletDriver #621: TabletDriver on Windows 10 20H1. Insider Preview Beta

Halozhan, 2020-04-12:
This issue has appeared after participating in Windows 10 Insider Program.
what's the problem: if "Smoothing filter" turns on, my tablet too slow or Polling rate around 80Hz.
So, I turned off this. A smoothing filter is necessary for me.
I know this project has been stopped 1~2 years. pls fix it.
IF WINDOWS 10 UPGRADE TO 20H1, MOST OF TABLETDRIVER USERS CAN'T USING SMOOTHING FILTER.

## Hardware/software information
Windows Version: Windows 10 20H1. Insider Preview Beta

Tablet Driver version: 0.2.4 - Self Compiled. others too

Tablet model: Wacom CTH-480

flubbateios, 2020-04-17:
I've also got this problem

Windows 10 Pro 19041.207 tested on 0.2.3 and Devocub fork.

CTL 480

Vlothx, 2020-05-07:
Does the same thing happens when using OpenTabletDriver filters?