hawku/TabletDriver #627: I have the same "Driver stopped. Restarting! Check console !!!" problems that people keep having and I've uninstalled and reinstalled my old wacom drivers, this driver,

elixxxirr, 2020-04-26:
I have the same "Driver stopped. Restarting! Check console !!!" problems that people keep having and I've uninstalled and reinstalled my old Wacom drivers, this driver, the vmulti-driver, and I've tried everything I read but it still won't work. 

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: Wacom CTL490

Vlothx, 2020-05-07:
Hello, looks like Hawku stopped the development of this driver. I would recommend trying [this fork](https://github.com/InfinityGhost/TabletDriver). You might need to disable your antivirus, or if don't want to do that, you can compile it by yourself.

usernamemadewheniplayedminecraft, 2020-05-11:
You might need to start tabletdriver as administrator