hawku/TabletDriver #630: Support for Wacom Intuos

LoowSs, 2020-04-28:
I'm new to git hub and don't really know where to ask but it would be awsome to have support for the wacom intuos.
Thanks!

Vlothx, 2020-05-07:
Uh...? Which Wacom Intuos you're talking about?

usernamemadewheniplayedminecraft, 2020-05-11:
The supported tablets are:
Supported tablets:
Wacom CTE-440
Wacom CTL-470
Wacom CTH-470
Wacom CTL-471
Wacom CTL-472
Wacom CTL-480
Wacom CTH-480
Wacom CTL-490
Wacom CTH-490
Wacom CTL-4100 USB
Wacom CTL-4100 Bluetooth
Wacom CTL-671
Wacom CTL-672
Wacom CTL-680
Wacom CTH-680
Wacom CTL-690
Wacom CTH-690
Wacom PTH-660
Wacom PTH-451
Wacom PTH-850
XP-Pen G430 (New 2017+ "Model B". Old G430 is not supported)
XP-Pen G430S
XP-Pen G540 Pro (G540 is not supported)
XP-Pen G640
XP-Pen G640S
XP-Pen Deco 01
XP-Pen Deco 01 v2
Huion 420
Huion H420
Huion H430P
Huion H640P
Huion osu!tablet
Gaomon S56K
VEIKK S640
Configured, but not properly tested:
Huion New 1060 Plus
Huion Inspiroy Q11K
Other Wacom tablets that might work: https://github.com/hawku/TabletDriver/blob/master/TabletDriverService/config/wacom.cfg