hawku/TabletDriver #634: Star g640 Tablet Not found!

airjordanlee, 2020-05-02:
I used to have this tablet a while back, along with this driver. However, whenever I run the program now, I can't get it to recognize my tablet. After deleting the XP-Pen PenTablet driver from device manager, I ran the program as administrator and was met with a VMulti not found error. I read the FAQ stating that downloading VMulti from PenTablet's package could solve it, but whenever I do that, I come back to the Tablet not found! error. I've been stuck in this loop for a while.

## Hardware/software information
Windows Version: 
10
Tablet Driver version: 
latest
Tablet model:
XP-Pen Star G640

Floffylg, 2020-05-04:
I have the exact same problem

JOJOx408, 2020-05-07:
I'm going to repost this because I recently bought the tablet and was gonna use the Hawku Drivers but as you know I can't.