hawku/TabletDriver #636: Can I use this for Turcom ?

Pleine, 2020-05-06:
I used to play osu with no problem on windows 7 in years ago with my Turcom ts-6580 and I haven't played the game since 5 years ago. But now I try to play the game again with my same old tablet my cursor shakes so much and cursor speed is not smooth (I mean can't control my cursor as will). The only driver that works for me back then and now is the one from their site and it just runs. No options or no configs. So there's nothing I can do even if I look for it I just can't find any way to fix it. But finally I found this custom driver and tried this but still no help. Is there any way to use this driver for my tablet ?

## Hardware/software information
Windows Version: wondows10x64

Tablet Driver version: ts-6580

Tablet model: Turcom

Vlothx, 2020-05-07:
I should note (that it looks like) Hawku stopped the development of this driver, so the possibilities of an update are very slim. Notheless, you might want to take a look at [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver), you can create your own configs there without too much hasle.

Pleine, 2020-05-08:
Thank you so much for your reply. Somehow I managed to change my tablet area with a lots of research from this site. Didn’t know this kind of such marvelous community exists until now. I will try which you mentioned and I’m sure this will help me a lot. Thanks again! Sent from Mail for Windows 10 From: VlothxSent: Friday, May 8, 2020 12:47 AMTo: hawku/TabletDriverCc: Pleine; AuthorSubject: Re: [hawku/TabletDriver] Can I use this for Turcom ? (#636) I should note (that it looks like) Hawku stopped the development of this driver, so the possibilities of an update are very slim. Notheless, you might want to take a look at OpenTabletDriver, you can create your own configs there without too much hasle.