hawku/TabletDriver #637: the cursor teleports

a0g-402, 2020-05-07:
https://youtu.be/g3aEVSPuHA4

## Hardware/software information
Windows Version: 10 Home

Tablet Driver version: TabletDriverV0.2.3

Tablet model: Wacom CTL-471

Vlothx, 2020-05-07:
Do you have Windows Ink enabled in the driver setting? If so, disable it/change it to Standard.

a0g-402, 2020-05-08:
> Do you have Windows Ink enabled in the driver setting? If so, disable it/change it to Standard.

no, I tried all filters too it still wont work

Vlothx, 2020-05-08:
Can you try if the same thing happens with [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver)?

a0g-402, 2020-05-09:
> Can you try if the same thing happens with [OpenTabletDriver](https://github.com/InfinityGhost/OpenTabletDriver)?

I tried, it doesn't work

usernamemadewheniplayedminecraft, 2020-05-11:
When you open opentabletdriver what does it say in the console with debug mode on?