hawku/TabletDriver #639: i can't install no driver at all and i still can't use it ctl472

wooo0061q, 2020-05-07:
## Hardware/software information
Windows Version: win10

Tablet Driver version: v2.3

Tablet model: ctl472
[image]
how do it do for this situation
[image]

Vlothx, 2020-05-07:
I would recommend trying [this fork](https://github.com/InfinityGhost/TabletDriver). You might need to disable your antivirus.

usernamemadewheniplayedminecraft, 2020-05-11:
You also need to start tablet driver as admin and if you end up using the fork Vlothx provided add an exclusion in your antivirus's settings

wooo0061q, 2020-05-12:
Thx dude ,the problem is solved,
 I appreciate it.

usernamethatimade4yearsagowheniplayedminecraft <notifications@github.com> 於
2020年5月12日 週二 上午2:39 寫道：