hawku/TabletDriver #642: DRIVER STOPPED, RESTARTING! CHECK CONSOLE!!

AFWAhab, 2020-05-14:
## Hardware/software information
Windows Version: 
Windows 10
Tablet Driver version: 
Latest
Tablet model: 
XP PEN G640S

Hawkus driver keeps saying driver stopped restarting, check console.. Idk what to do man, this is what the startuplog says:
[2020-05-14 18:45:49.182] \ Reading 'config/init.cfg'
[2020-05-14 18:45:49.182] >> Log startuplog.txt
[2020-05-14 18:45:49.182] Log file 'startuplog.txt' opened.
[2020-05-14 18:45:49.182] >> LogDirect true
[2020-05-14 18:45:49.182] Log direct print = True
[2020-05-14 18:45:49.182] >> Include "config\tablet.cfg"
[2020-05-14 18:45:49.183] \ Reading 'config\tablet.cfg'
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> # Example tablet definition:
[2020-05-14 18:45:49.183] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> # VID: 0x056a
[2020-05-14 18:45:49.183] >> # PID: 0x00dd
[2020-05-14 18:45:49.183] >> # HID Usage Page: 0x0D
[2020-05-14 18:45:49.183] >> # HID Usage: 0x01
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> # XP-Pen G640
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> Tablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}" 6 G640
[2020-05-14 18:45:49.183] >> Name "XP-Pen G640"
[2020-05-14 18:45:49.183] >> ReportLength 8
[2020-05-14 18:45:49.183] >> ButtonMask 0x80
[2020-05-14 18:45:49.183] >> MaxX 8340
[2020-05-14 18:45:49.183] >> MaxY 4680
[2020-05-14 18:45:49.183] >> MaxPressure 2047
[2020-05-14 18:45:49.183] >> Width 101.6
[2020-05-14 18:45:49.183] >> Height 56.6
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> # Wacom CTL-470 (Wacom drivers installed)
[2020-05-14 18:45:49.183] >> #
[2020-05-14 18:45:49.183] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2020-05-14 18:45:49.184] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2020-05-14 18:45:49.184] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2020-05-14 18:45:49.184] >> ReportId 0x02
[2020-05-14 18:45:49.184] >> ReportLength 11
[2020-05-14 18:45:49.184] >> DetectMask 0x40
[2020-05-14 18:45:49.184] >> MaxX 14720
[2020-05-14 18:45:49.184] >> MaxY 9200
[2020-05-14 18:45:49.185] >> MaxPressure 1023
[2020-05-14 18:45:49.185] >> Width 147.2
[2020-05-14 18:45:49.185] >> Height 92.0
[2020-05-14 18:45:49.185] >> TabletFormat WacomDrivers
[2020-05-14 18:45:49.185] >> #
[2020-05-14 18:45:49.185] >> # Wacom CTL-470 
[2020-05-14 18:45:49.185] >> #
[2020-05-14 18:45:49.185] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2020-05-14 18:45:49.186] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-14 18:45:49.186] >> Name "Wacom CTL-470"
[2020-05-14 18:45:49.186] >> ReportId 0x02
[2020-05-14 18:45:49.186] >> ReportLength 10
[2020-05-14 18:45:49.186] >> DetectMask 0x40
[2020-05-14 18:45:49.186] >> MaxX 14720
[2020-05-14 18:45:49.186] >> MaxY 9200
[2020-05-14 18:45:49.186] >> MaxPressure 1023
[2020-05-14 18:45:49.186] >> Width 147.2
[2020-05-14 18:45:49.186] >> Height 92.0
[2020-05-14 18:45:49.186] >> InitFeature 0x02 0x02
[2020-05-14 18:45:49.186] >> #
[2020-05-14 18:45:49.186] >> # Wacom CTH-470 (Wacom drivers installed)
[2020-05-14 18:45:49.186] >> #
[... 1511 lines cut ...]
[2020-05-14 18:45:49.413] [STATUS] COMMANDS AuxReportId AuxReportLength Benchmark Bench ButtonMap AuxButtonMap AuxButtons Buttons 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2020-05-14 18:45:49.413] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2020-05-14 18:45:49.413] 
[2020-05-14 18:45:49.413]   Driver version: 0.2.3
[2020-05-14 18:45:49.413]   Windows version: Microsoft Windows NT 6.2.9200.0
[2020-05-14 18:45:49.413]   Windows product: Windows 10 Enterprise
[2020-05-14 18:45:49.413]   Windows release: 1809
[2020-05-14 18:45:49.413] 
[2020-05-14 18:45:49.413] >> CheckTablet
[2020-05-14 18:45:49.413] [ERROR] Tablet not found!
[2020-05-14 18:45:49.413] [ERROR] Check the list of supported tablets from the GitHub page.
[2020-05-14 18:45:49.413] [ERROR] http://github.com/hawku/TabletDriver

iDK WHAT to do man

harryxing, 2020-05-15:
I have the same problem and I don't know what to do either

Tazymax, 2020-05-16:
Same problem but on xp pen 430s, tryed a bunch of options but nothing worked

ssccoonniicc, 2020-05-17:
Did you try starting as administrator?