hawku/TabletDriver #647: XP-PEN Star G640 Recognized As "UGTABLET"

Sxuccy, 2020-05-19:
I recently got the xp pen star g640 and the tablet isnt working getting detected by the drivers which is strange because it says tablet found in console but then where it says " Device string (Manufacture) 'UGTABLET' does not match with 'XP-PEN' .  Tablet Invalid" Can someone help me pls

[image]

[2020-05-19 04:55:22.968] >> Width 152.000
[2020-05-19 04:55:22.968] >> Height 95.000
[2020-05-19 04:55:22.968] >> InitFeature 0x02 0x02
[2020-05-19 04:55:22.968] >> # Wacom CTH-490
[2020-05-19 04:55:22.968] >> HIDTablet 0x056A 0x033C 0xFF0D 0x0001
[2020-05-19 04:55:23.000] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x033C UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.000] >> HIDTablet 0x056A 0x033C 0x000D 0x0001
[2020-05-19 04:55:23.031] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x033C UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.031] >> Name "Wacom CTH-490"
[2020-05-19 04:55:23.031] >> ReportLength 10
[2020-05-19 04:55:23.031] >> DetectMask 0x40
[2020-05-19 04:55:23.031] >> MaxX 15200
[2020-05-19 04:55:23.031] >> MaxY 9500
[2020-05-19 04:55:23.031] >> MaxPressure 2047
[2020-05-19 04:55:23.031] >> Width 152.000
[2020-05-19 04:55:23.031] >> Height 95.000
[2020-05-19 04:55:23.031] >> InitFeature 0x02 0x02
[2020-05-19 04:55:23.031] >> TabletFormat WacomIntuosV2
[2020-05-19 04:55:23.031] >> # Wacom CTH-661
[2020-05-19 04:55:23.031] >> HIDTablet 0x056A 0x00D3 0xFF0D 0x0001
[2020-05-19 04:55:23.064] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D3 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.064] >> HIDTablet 0x056A 0x00D3 0x000D 0x0001
[2020-05-19 04:55:23.096] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D3 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.096] >> Name "Wacom CTH-661"
[2020-05-19 04:55:23.096] >> ReportLength 9
[2020-05-19 04:55:23.096] >> DetectMask 0x40
[2020-05-19 04:55:23.096] >> MaxX 21648
[2020-05-19 04:55:23.096] >> MaxY 13700
[2020-05-19 04:55:23.096] >> MaxPressure 1023
[2020-05-19 04:55:23.096] >> Width 216.480
[2020-05-19 04:55:23.096] >> Height 137.000
[2020-05-19 04:55:23.096] >> InitFeature 0x02 0x02
[2020-05-19 04:55:23.096] >> # Wacom CTH-661(A)
[2020-05-19 04:55:23.096] >> HIDTablet 0x056A 0x00D8 0xFF0D 0x0001
[2020-05-19 04:55:23.128] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D8 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.128] >> HIDTablet 0x056A 0x00D8 0x000D 0x0001
[2020-05-19 04:55:23.160] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D8 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.160] >> Name "Wacom CTH-661(A)"
[2020-05-19 04:55:23.160] >> ReportLength 9
[2020-05-19 04:55:23.160] >> DetectMask 0x40
[2020-05-19 04:55:23.160] >> MaxX 21648
[2020-05-19 04:55:23.160] >> MaxY 13700
[2020-05-19 04:55:23.160] >> MaxPressure 1023
[2020-05-19 04:55:23.160] >> Width 216.480
[2020-05-19 04:55:23.160] >> Height 137.000
[2020-05-19 04:55:23.160] >> InitFeature 0x02 0x02
[2020-05-19 04:55:23.160] >> # Wacom CTH-661SE
[2020-05-19 04:55:23.160] >> HIDTablet 0x056A 0x00DB 0xFF0D 0x0001
[2020-05-19 04:55:23.193] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DB UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.193] >> HIDTablet 0x056A 0x00DB 0x000D 0x0001
[2020-05-19 04:55:23.224] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DB UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.225] >> Name "Wacom CTH-661SE"
[2020-05-19 04:55:23.225] >> ReportLength 9
[2020-05-19 04:55:23.225] >> DetectMask 0x40
[2020-05-19 04:55:23.225] >> MaxX 21648
[2020-05-19 04:55:23.225] >> MaxY 13700
[2020-05-19 04:55:23.225] >> MaxPressure 1023
[2020-05-19 04:55:23.225] >> Width 216.480
[2020-05-19 04:55:23.225] >> Height 137.000
[2020-05-19 04:55:23.225] >> InitFeature 0x02 0x02
[2020-05-19 04:55:23.225] >> # Wacom CTH-670
[2020-05-19 04:55:23.225] >> HIDTablet 0x056A 0x00DF 0xFF0D 0x0001
[2020-05-19 04:55:23.257] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DF UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.257] >> HIDTablet 0x056A 0x00DF 0x000D 0x0001
[2020-05-19 04:55:23.289] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DF UsagePage=0x000D Usage=0x0001 Exclusive=False
[2020-05-19 04:55:23.289] >> Name "Wacom CTH-670"
[... 209 lines cut ...]
[2020-05-19 04:55:23.972] [STATUS] COMMANDS AntiSmoothingFilter Anti AntiSmoothing AuxButtonCount AuxDetectMask AuxHID AuxIgnoreMask 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS AuxReportId AuxReportLength Benchmark Bench ButtonMap AuxButtonMap AuxButtons Buttons 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2020-05-19 04:55:23.972] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2020-05-19 04:55:23.972] 
[2020-05-19 04:55:23.972]   Driver version: 0.2.3
[2020-05-19 04:55:23.972]   Windows version: Microsoft Windows NT 6.2.9200.0
[2020-05-19 04:55:23.972]   Windows product: Windows 10 Home
[2020-05-19 04:55:23.972]   Windows release: 1809
[2020-05-19 04:55:23.972] 
[2020-05-19 04:55:23.972] >> CheckTablet
[2020-05-19 04:55:23.972] [ERROR] Tablet not found!
[2020-05-19 04:55:23.972] [ERROR] Check the list of supported tablets from the GitHub page.
[2020-05-19 04:55:23.972] [ERROR] http://github.com/hawku/TabletDriver
Driver stopped!

Sxuccy, 2020-05-19:
Also I checked the config files and theres nothing I found that nothing that has UGTABLET as the check string manufacture

meowzy0, 2020-05-19:
oh i so happy, configs from this helped me https://github.com/InfinityGhost/TabletDriver

meowzy0, 2020-05-19:
idk how

Sxuccy, 2020-05-20:
@meowzy0 so hawku works for you now?

meowzy0, 2020-05-20:
i think yes, but i dont see ant difference between xp pen drivers and hawku, maybe, but smoothing , anti-smoothing etc. is working

meowzy0, 2020-05-20:
[image]

meowzy0, 2020-05-20:
this what i found in config

meowzy0, 2020-05-20:
maybe try to rename checkstring product to 6 inch PenTablet

Sxuccy, 2020-05-21:
I think the problem is that if you look at it again under where it says tablet found, the driver is reading the manufacture from the tablet as "UGTABLET" I looked through the configs and did research. "UGTABLET" has nothing to do with my tablet so I dont know anymore mann lol.

[2020-05-19 04:55:23.705] Tablet found! Manufacturer='UGTABLET', Product='6 inch PenTablet', Serial='000000'
[2020-05-19 04:55:23.705] >> CheckString Manufacturer "XP-PEN"
[2020-05-19 04:55:23.705] Device string (Manufacturer) 'UGTABLET' does not match with 'XP-PEN'. Tablet invalid!

Vlothx, 2020-06-08:
Comment (or delete) the CheckString line.

mmdorik28-spec, 2025-11-20:
i have mac and connected my deco mini 7w in system infomation i found Manufacturer UGTABLET 

[image]