hawku/TabletDriver #664: Smoothing filter lag in Windows 10 ver 2004

Alegzandr, 2020-05-30:
Hi, 

Since I updated to Windows 10 ver 2004, whenever I enable smoothing filter in order to use 1000hz polling rate, mouse cursor seems like 60hz on my 240hz display, even if I use another latency or another rate.

How can it be fixed ? It's unplayable for me

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

ssccoonniicc, 2020-05-30:
TabletDriver’s Smoothing is pretty broken in newer windows builds it seems. I would recommend using [InfinityGhost’s OpenTabletDriver](github.com/InfinityGhost/OpenTabletDriver) . Just download the release and plugin dll for it.

Alegzandr, 2020-05-30:
Sorry but there is no 1000hz and no antichatter in your driver ssccoonniicc

AbdShullah, 2020-05-30:
My devocub fork fixes that issue: https://github.com/AbdShullah/TabletDriver/releases note that cpu usage can be higher, but it shouldn't lag the game.

Alegzandr, 2020-05-31:
Thank you so much !

Leector, 2020-06-14:
Didn't worked for me :/